NVIDIA-AI-IOT/Lidar_AI_Solution #128: Can not find any matched kernel 9 x 32

akjt, 2023-07-14:
Hi,

We are unfortunately blocked by the following error: `Assert failed 💀. false in file src/spconv/implicit-gemm.cu:383, message: Can not find any matched kernel 9 x 32
`
I understand that you only support Nx4/Nx5/Nx8/Nx16 . The question is if you would be able to  support Nx32 or allow us to add this extension?

hopef, 2023-07-14:
Currently, Nx32 and Nx16 are supported.

akjt, 2023-07-14:
since when was Nx32 supported. I am running the most recent master branch and I get the error above which suggests that Nx32 is not supported?

hopef, 2023-07-14:
The supported shape pairs are called MxN, which means the number of input channels is M, and the number of output channels is N.

These support shape pairs are listed below:
128x128
16x16
16x32
32x32
32x64
4x16
5x16
64x64
8x16

For your committed error message. It has 9 input channels and 32 output channels. That's an unsupported pattern on the above list.

akjt, 2023-07-14:
Okay thanks for the clarification. Any plans to extend the above  list for more shape pairs?

hopef, 2023-07-14:
We will let you know if we have any updates.

hopef, 2023-12-07:
or any updates to whether you would open source the source code behind libspconv?
-> Currently, we have no plans to open-source the libspconv code.

Is the src code similar to traveller59/spconv github repo?
-> Totally different with traveller59/spconv, src code is based on CUDA kernels(TensorCore only) instead of CUTLASS.
-> We developed the highly optimized engine(contains Network build, low-level op, memory reuse, and layerfuse ...), But traveller59/spconv just provides a low-level op implementation for training, not for inference.