NVIDIA-AI-IOT/Lidar_AI_Solution #129: How to convert pcd file or ros point_cloud msg for CUDA_bevfusion in c++?

lucianzhong, 2023-07-17:
How to convert pcd file or ros point_cloud msg for CUDA_bevfusion in c++?
Could you please provide a demo on using loadbinary() which is used to load a pcd file right ?

Best regards

lucianzhong, 2023-07-17:
below is ros subscribe code. The nv::Tensor::from_data_reference() part is wrong. Could you point out how to make it right?

void cloud_cb(const sensor_msgs::PointCloud2::ConstPtr &msg)
{
    if (!has_init)
    {
        pcl::PointCloud<pcl::PointXYZI>::Ptr ROI_cloud(new pcl::PointCloud<pcl::PointXYZI>);
        pcl::fromROSMsg(*msg, *ROI_cloud);
        std::cout << "ROI_cloud->points.size()   " << ROI_cloud->points.size() << std::endl;
        float *points = new float[ROI_cloud->points.size()];
        for (int i = 0; i < ROI_cloud->points.size(); i++)
        {
            points[i + 0] = ROI_cloud->points[i].x;
            points[i + 1] = ROI_cloud->points[i].y;
            points[i + 2] = ROI_cloud->points[i].z;
        }
        void *pc_data = static_cast<void *>(points);

        vector<int32_t> shape{ROI_cloud->points.size()};
        // Tensor Tensor::from_data_reference(void *data, vector<int32_t> shape, DataType dtype, bool device)
        lidar_point_cloud_tensor = nv::Tensor::from_data_reference(pc_data, shape, nv::DataType::Float16, false);

        std::cout << "cloud_cb" << std::endl;
        has_init = true;
    }
}

hopef, 2023-07-17:
```c++
typedef unsigned short half;

static inline half __internal_float2half(const float f)
{
    unsigned int x;
    unsigned int u;
    unsigned int result;
    (void)memcpy(&x, &f, sizeof(f));
    u = (x & 0x7fffffffU);
    sign = ((x >> 16U) & 0x8000U);
    // NaN/+Inf/-Inf
    if (u >= 0x7f800000U) {
        remainder = 0U;
        result = ((u == 0x7f800000U) ? (sign | 0x7c00U) : 0x7fffU);
    } else if (u > 0x477fefffU) { // Overflows
        remainder = 0x80000000U;
        result = (sign | 0x7bffU);
    } else if (u >= 0x38800000U) { // Normal numbers
        remainder = u << 19U;
        u -= 0x38000000U;
        result = (sign | (u >> 13U));
    } else if (u < 0x33000001U) { // +0/-0
        remainder = u;
        result = sign;
    } else { // Denormal numbers
        const unsigned int exponent = u >> 23U;
        const unsigned int shift = 0x7eU - exponent;
        unsigned int mantissa = (u & 0x7fffffU);
        mantissa |= 0x800000U;
        remainder = mantissa << (32U - shift);
        result = (sign | (mantissa >> shift));
        result &= 0x0000FFFFU;
    }
    unsigned short x = static_cast<unsigned short>(result);
    if ((remainder > 0x80000000U) || ((remainder == 0x80000000U) && ((x & 0x1U) != 0U))) {
        x++;
    }
    return x;
}

void cloud_cb(const sensor_msgs::PointCloud2::ConstPtr &msg)
{
    pcl::PointCloudpcl::PointXYZI::Ptr ROI_cloud(new pcl::PointCloudpcl::PointXYZI);
    pcl::fromROSMsg(*msg, *ROI_cloud);
    std::cout << "ROI_cloud->points.size() " << ROI_cloud->points.size() << std::endl;
    half *points = new half[ROI_cloud->points.size() * 3];
    for (int i = 0; i < ROI_cloud->points.size(); i++)
    {
        points[i * 3 + 0] = __internal_float2half(ROI_cloud->points[i].x);
        points[i * 3 + 1] = __internal_float2half(ROI_cloud->points[i].y);
        points[i * 3 + 2] = __internal_float2half(ROI_cloud->points[i].z);
    }
    vector<int32_t> shape{ROI_cloud->points.size(), 3};
    // Tensor Tensor::from_data_reference(void *data, vector<int32_t> shape, DataType dtype, bool device)
    lidar_point_cloud_tensor = nv::Tensor::from_data_reference(points, shape, nv::DataType::Float16, false);
}
```

And you should refer the issue #124.

lucianzhong, 2023-07-18:
Thanks a lot for the replay. Be of great help for me.  The usable code is posted below.

lucianzhong, 2023-07-18:
typedef unsigned short half;
static inline half __internal_float2half(const float f)
{
    unsigned int x;
    unsigned int u;
    unsigned int result;
    unsigned int sign;
    unsigned int remainder;
    (void)memcpy(&x, &f, sizeof(f));
    u = (x & 0x7fffffffU);
    sign = ((x >> 16U) & 0x8000U);
    // NaN/+Inf/-Inf
    if (u >= 0x7f800000U)
    {
        remainder = 0U;
        result = ((u == 0x7f800000U) ? (sign | 0x7c00U) : 0x7fffU);
    }
    else if (u > 0x477fefffU)
    { // Overflows
        remainder = 0x80000000U;
        result = (sign | 0x7bffU);
    }
    else if (u >= 0x38800000U)
    { // Normal numbers
        remainder = u << 19U;
        u -= 0x38000000U;
        result = (sign | (u >> 13U));
    }
    else if (u < 0x33000001U)
    { // +0/-0
        remainder = u;
        result = sign;
    }
    else
    { // Denormal numbers
        const unsigned int exponent = u >> 23U;
        const unsigned int shift = 0x7eU - exponent;
        unsigned int mantissa = (u & 0x7fffffU);
        mantissa |= 0x800000U;
        remainder = mantissa << (32U - shift);
        result = (sign | (mantissa >> shift));
        result &= 0x0000FFFFU;
    }

    unsigned short x_tmp = static_cast<unsigned short>(result);
    if ((remainder > 0x80000000U) || ((remainder == 0x80000000U) && ((x & 0x1U) != 0U)))
    {
        x_tmp++;
    }
    return x_tmp;
}

void cloud_cb(const sensor_msgs::PointCloud2::ConstPtr &msg)
{
    pcl::PointCloud<pcl::PointXYZI>::Ptr ROI_cloud(new pcl::PointCloud<pcl::PointXYZI>);
    pcl::fromROSMsg(*msg, *ROI_cloud);
    std::cout << "ROI_cloud->points.size()   " << ROI_cloud->points.size() << std::endl;
    half *points = new half[ROI_cloud->points.size() * 5];
    for (int i = 0; i < ROI_cloud->points.size(); i++)
    {
        points[i * 5 + 0] = __internal_float2half(ROI_cloud->points[i].x);
        points[i * 5 + 1] = __internal_float2half(ROI_cloud->points[i].y);
        points[i * 5 + 2] = __internal_float2half(ROI_cloud->points[i].z);
        points[i * 5 + 3] = __internal_float2half(1);
        points[i * 5 + 4] = __internal_float2half(0);
    }
    vector<int32_t> shape{ROI_cloud->points.size(), 5};
    // Tensor Tensor::from_data_reference(void *data, vector<int32_t> shape, DataType dtype, bool device)
    lidar_point_cloud_tensor = nv::Tensor::from_data_reference(points, shape, nv::DataType::Float16, false);
}

kunnalparihar, 2023-08-18:
@lucianzhong @hopef 
I am trying to add pcl library to CMakeLists.txt using "find_package (PCL 1.10 REQUIRED)"  but when I compile and build the code it shows below error. Can you help me with this? 
--------------------------------------------------------------------
[  4%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/common/bevfusion_core_generated_tensor.cu.o
<command-line>: error: ISO C++11 requires whitespace after the macro name [-Werror]
cc1plus: all warnings being treated as errors
CMake Error at bevfusion_core_generated_tensor.cu.o.Release.cmake:220 (message):
  Error generating
  /root/workspace/new_test_lidar_ai_solution_working/Lidar_AI_Solution/CUDA-BEVFusion/build/CMakeFiles/bevfusion_core.dir/src/common/./bevfusion_core_generated_tensor.cu.omake[2]: *** [CMakeFiles/bevfusion_core.dir/build.make:140: CMakeFiles/bevfusion_core.dir/src/common/bevfusion_core_generated_tensor.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:86: CMakeFiles/bevfusion_core.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
-------------------------------------------------

lucianzhong, 2023-08-20:
it is related to g++ version,please check.
or you can manually modify your code like from origin "UpdateDeploy platetime[%"PRI64"]"  to  "UpdateDeploy starttime[%" PRI64 "]"

Gigalomanicx, 2023-08-28:
> @lucianzhong @hopef I am trying to add pcl library to CMakeLists.txt using "find_package (PCL 1.10 REQUIRED)" but when I compile and build the code it shows below error. Can you help me with this?
> 
> [ 4%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/common/bevfusion_core_generated_tensor.cu.o : error: ISO C++11 requires whitespace after the macro name [-Werror] cc1plus: all warnings being treated as errors CMake Error at bevfusion_core_generated_tensor.cu.o.Release.cmake:220 (message):   Error generating   /root/workspace/new_test_lidar_ai_solution_working/Lidar_AI_Solution/CUDA-BEVFusion/build/CMakeFiles/bevfusion_core.dir/src/common/./bevfusion_core_generated_tensor.cu.omake[2]: *** [CMakeFiles/bevfusion_core.dir/build.make:140: CMakeFiles/bevfusion_core.dir/src/common/bevfusion_core_generated_tensor.cu.o] Error 1 make[1]: *** [CMakeFiles/Makefile2:86: CMakeFiles/bevfusion_core.dir/all] Error 2 make: *** [Makefile:91: all] Error 2

I encountered the same problem.

hopef, 2023-08-28:
Try to remove the `-Wall` flags:

CUDA-BEVFusion/CMakeLists.txt
```CMake
set(CMAKE_CXX_FLAGS_RELEASE "-std=c++14 -Wextra -Wall -Wno-missing-field-initializers -Wno-deprecated-declarations -O3 -DENABLE_TEXT_BACKEND_STB")
set(CMAKE_CXX_FLAGS_DEBUG   "-std=c++14 -O0 -g -DENABLE_TEXT_BACKEND_STB")

set(CUDA_NVCC_FLAGS_RELEASE "-Werror=all-warnings -Xcompiler -std=c++14,-Wextra,-Wall,-Wno-deprecated-declarations,-O3 -DENABLE_TEXT_BACKEND_STB")
set(CUDA_NVCC_FLAGS_DEBUG   "-std=c++14 -O0 -g -DENABLE_TEXT_BACKEND_STB")
```

to
```CMake
set(CMAKE_CXX_FLAGS_RELEASE "-std=c++14 -Wno-missing-field-initializers -Wno-deprecated-declarations -O3 -DENABLE_TEXT_BACKEND_STB")
set(CMAKE_CXX_FLAGS_DEBUG   "-std=c++14 -O0 -g -DENABLE_TEXT_BACKEND_STB")

set(CUDA_NVCC_FLAGS_RELEASE "-Xcompiler -std=c++14,-Wno-deprecated-declarations,-O3 -DENABLE_TEXT_BACKEND_STB")
set(CUDA_NVCC_FLAGS_DEBUG   "-std=c++14 -O0 -g -DENABLE_TEXT_BACKEND_STB")
```

Gigalomanicx, 2023-08-28:
It works for me, Thanks !

linClubs, 2023-09-11:
Hi, I want to used OpenCV read image, how to  convert OpenCV  image  Mat object for CUDA_bevfusion in c++?   Thanks