NVIDIA-AI-IOT/Lidar_AI_Solution #131: A question about pillarScatter kernel function may have bank conflict

usergxx, 2023-07-20:
I don't understand why pillarScatter impleted like this.
First,copy the pillar feature into the shared memory and then from shared memory copy to spatial_feature_data.
In the second stage, from shared memory copy to spatial_feature_data have bank conflict.
`spatial_feature_data[i*featureY*featureX + y*featureX + x] = pillarSM[threadIdx.x][i];`

why not direct copy the spatial_feature_data from pillar_features_data. like this
```
__global__ void pillarScatterFloatkernel(const float *pillar_features_data,
                                         const unsigned int *coords_data, const unsigned int *params_data,
                                         unsigned int featureX, unsigned int featureY,
                                         float *spatial_feature_data)
{
    int pillar_idx = blockIdx.x * PILLARS_PER_BLOCK + threadIdx.x;
    const int num_pillars = params_data[0];
    if(pillar_idx >= num_pillars) return;
    uint4 coord = ((const uint4 *)coords_data)[pillar_idx];
    unsigned int x = coord.w;
    unsigned int y = coord.z;

    for (int i = 0; i < PILLAR_FEATURE_SIZE; i++)
    {
        spatial_feature_data[i*featureY*featureX + y*featureX + x] = pillar_features_data[(blockIdx.x * PILLARS_PER_BLOCK +threadIdx.x)*PILLAR_FEATURE_SIZE + i];
    }
}
```
I am looking forward to your response. Thank you

hopef, 2023-07-21:
Maybe you can compare them.
It seems to optimize the performance of transpose.

usergxx, 2023-07-21:
> 

i will try it, thank you for you reply. I have one more question for you
In CUDA-PointPillars project  preprocess_kernels.cu   generateFeatures_launch  function
```
dim3 blocks( (MAX_VOXELS+WARPS_PER_BLOCK-1)/WARPS_PER_BLOCK);
dim3 threads(WARPS_PER_BLOCK*WARP_SIZE);
generateFeatures_kernel<<<blocks, threads, 0, stream>>>
  (voxel_features,
  voxel_num,
  voxel_idxs,
  params,
  voxel_x, voxel_y, voxel_z,
  range_min_x, range_min_y, range_min_z,
  features);
```
It avoid  block.x to large, so divide by WARPS_PER_BLOCK， the WARPS_PER_BLOCK is 4.
But the threads.x is WARPS_PER_BLOCK*WARP_SIZE, WARP_SIZE is 32. This also will lead to a small bank conflict.
I think writing it like this would be better, set WARP_SIZE to blocks.y
```
dim3 blocks( (MAX_VOXELS+WARPS_PER_BLOCK-1)/WARPS_PER_BLOCK, WARPS_PER_BLOCK);
dim3 threads(WARP_SIZE);
generateFeatures_kernel<<<blocks, threads, 0, stream>>>
  (voxel_features,
  voxel_num,
  voxel_idxs,
  params,
  voxel_x, voxel_y, voxel_z,
  range_min_x, range_min_y, range_min_z,
  features);
```
Is it like this?

closed 2023-07-21 by usergxx