NVIDIA-AI-IOT/Lidar_AI_Solution #132: Cannot create TRT engine from scn.onnx because of BatchNorm

qqcx, 2023-07-21:
Hello author, I am a Chinese. Thank you for your code.
I tried to export scn.onnx to trt, then this happened.
In CenterPoint code, BatchNorm is built by function: build_norm_layer.

The log of bug:
[Verb🚩] Parse node bn0 [BatchNorm1d], numInput=1, numOutput=1
Assert failed 💀. false in file src/spconv/engine.cu:1528, message: Unsupport operator [BatchNorm1d]
Aborted (core dumped)

hopef, 2023-07-21:
You should fuse the BatchNorm1d with the previous SparseConv layer.
The `libspconv.so` implements spconv&bais instead of two separate operators(spconv and `batchnorm`) to improve performance.

hopef, 2023-07-21:
You can refer to [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/ca3acfe07310d795fc9c3e75df1d1bf0bc78e461/libraries/3DSparseConvolution/tool/bevfusion-export/export-scn.py#L79C33-L79C33).

qqcx, 2023-07-27:
That helps a lot!
This works:  model = funcs.layer_fusion(model)

closed 2023-08-08 by qqcx