NVIDIA-AI-IOT/Lidar_AI_Solution #135: 'SwinTransformer' object has no attribute 'layer3'

kunnalparihar, 2023-07-24:
Hii,
Thanks for the great work. I want to convert bevfusion swinT+voxelnet model to onnx and evaluate the performance.
I trained my bevfusion model with bevfusion/configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml config and generated checkpoint file epoch.pth.
Now, I am trying to quantize epoch.pth using
$ python qat/ptq.py --config=bevfusion/configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --ckpt=/usr/src/app/ws/data/kunal_data/nvidia_bevfusion_data/model/camlid_k/epoch_6.pth --calibrate_batch 300

But it shows this below error,
load checkpoint from local path: /usr/src/app/ws/data/kunal_data/nvidia_bevfusion_data/model/camlid_k/epoch_6.pth
---type model_camera_branch.backbone  <class 'mmdet.models.backbones.swin.SwinTransformer'>
Traceback (most recent call last):
  File "qat/ptq.py", line 150, in <module>
    main()
  File "qat/ptq.py", line 130, in main
    model = quantize_net(model)
  File "qat/ptq.py", line 81, in quantize_net
    quantize.quantize_encoders_camera_branch(model.encoders.camera)
  File "/usr/src/app/ws/Lidar_AI_Solution/CUDA-BEVFusion/qat/lean/quantize.py", line 176, in quantize_encoders_camera_branch
    major = model_camera_branch.backbone.layer3[0].conv1._input_quantizer
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1177, in __getattr__
    raise AttributeError("'{}' object has no attribute '{}'".format(
AttributeError: 'SwinTransformer' object has no attribute 'layer3'

NOTE: I tried converting resnet50 checkpoint file to onnx and it works fine but swinT does not work.
Can you help me with this?

hopef, 2023-07-25:
The quantize.py is mainly to improve the ResNet50-PTQ performance.
It would be best if you disabled the PTQ for the SwinT backbone.

sandeepnmenon, 2023-07-26:
Can we run the `export-camera.py` etc scripts to export the onnx modules from a non quantized model?
Would those onnx models work in the inference pipeline of bevfusion.cpp just slower?

cdefg, 2023-07-31:
@sandeepnmenon @hopef  The Swint model (or other models) from bevfusion's original repo is type of single .pth file(follow their readme training step), how can I do to export the  onnx  models (model/fuser.onnx, etc) from training steps ?  

And I found if I use ptq.py to export swint model, failure from this issue will happen.

sandeepnmenon, 2023-07-31:
@cdefg 
You can use the original repo's model to run the export scripts and each script extracts out the components like model/fuser.onnx, the camera backbone etc.

The `ptq.py` is not written for the swint model. It only works with the resnet backbone as mentioned above in this issue

cdefg, 2023-08-01:
> 

Thank you for your great idea. 
I have tried to export onnx using bevfusion/tools/export.py and modify it to adapt for onnx output, but still got failure:

```Segmentation fault (core dumped)```

The `export.py` code is :

```
import argparse
import os
import time
import warnings

import mmcv
import onnx
import torch
from mmcv import Config, DictAction
from mmcv.parallel import MMDataParallel, MMDistributedDataParallel
from mmcv.runner import get_dist_info, init_dist, load_checkpoint, wrap_fp16_model
from mmdet3d.apis import single_gpu_test
from mmdet3d.datasets import build_dataloader, build_dataset
from mmdet3d.models import build_model
from mmdet.apis import multi_gpu_test, set_random_seed
from mmdet.datasets import replace_ImageToTensor
from onnxsim import simplify
from tqdm import tqdm

def parse_args():
    parser = argparse.ArgumentParser(description="MMDet test (and eval) a model")
    parser.add_argument("config", help="test config file path")
    parser.add_argument("checkpoint", help="checkpoint file")
    parser.add_argument(
        "--cfg-options",
        nargs="+",
        action=DictAction,
        help="override some settings in the used config, the key-value pair "
        "in xxx=yyy format will be merged into config file. If the value to "
        'be overwritten is a list, it should be like key="[a,b]" or key=a,b '
        'It also allows nested list/tuple values, e.g. key="[(a,b),(c,d)]" '
        "Note that the quotation marks are necessary and that no white space "
        "is allowed.",
    )
    args = parser.parse_args()
    return args

def main():
    args = parse_args()

    cfg = Config.fromfile(args.config)
    if args.cfg_options is not None:
        cfg.merge_from_dict(args.cfg_options)

    # set cudnn_benchmark
    torch.backends.cudnn.benchmark = True

    cfg.model.pretrained = None
    print(cfg)
    if isinstance(cfg.data.test, dict):
        cfg.data.test.test_mode = True
    elif isinstance(cfg.data.test, list):
        for ds_cfg in cfg.data.test:
            ds_cfg.test_mode = True

    # build the dataloader
    dataset = build_dataset(cfg.data.test)
    data_loader = build_dataloader(
        dataset,
        samples_per_gpu=1,
        workers_per_gpu=cfg.data.workers_per_gpu,
        dist=False,
        shuffle=False,
    )

    # build the model and load checkpoint
    cfg.model.train_cfg = None
    model = build_model(cfg.model, test_cfg=cfg.get("test_cfg"))
    checkpoint = load_checkpoint(model, args.checkpoint, map_location="cpu")

    # old versions did not save class info in checkpoints, this walkaround is
    # for backward compatibility
    if "CLASSES" in checkpoint.get("meta", {}):
        model.CLASSES = checkpoint["meta"]["CLASSES"]
    else:
        model.CLASSES = dataset.CLASSES

    model.eval()

    with torch.no_grad():
        for data in data_loader:
            img = torch.cat(data["img"].data, dim=0)
            points = data["points"].data[0]
            camera2ego = data["camera2ego"].data[0]
            lidar2ego = data["lidar2ego"].data[0]
            lidar2camera = data["lidar2camera"].data[0]
            lidar2image = data["lidar2image"].data[0]
            camera_intrinsics = data["camera_intrinsics"].data[0]
            camera2lidar = data["camera2lidar"].data[0]
            img_aug_matrix = data["img_aug_matrix"].data[0]
            lidar_aug_matrix = data["lidar_aug_matrix"].data[0]

            data_input = (
                img,
                points,
                camera2ego,
                lidar2ego,
                lidar2camera,
                lidar2image,
                camera_intrinsics,
                camera2lidar,
                img_aug_matrix,
                lidar_aug_matrix,
            )
            print(data_input)
            print(img.shape)

            # print(img.shape)
            metas = data["metas"].data

            from functools import partial

            model.forward = partial(
                model.forward,
                metas=metas,
                rescale=True,
            )

            torch.onnx.export(
                model,
                data_input,
                "model.onnx",
                input_names=("img", "points", "camera2ego", "lidar2ego", "lidar2camera", "lidar2image", "camera_intrinsics", "camera2lidar", "img_aug_matrix", "lidar_aug_matrix"),
                opset_version=13,
                do_constant_folding=True,
            )
            model = onnx.load("model.onnx")
            model, _ = simplify(model)
            onnx.save(model, "model.onnx")
            return

if __name__ == "__main__":
    main()

``` 
And the script I used is 

```
python tools/export.py runs/run-490c9f75/configs.yaml runs/run-490c9f75/epoch_20.pth
```

still need help, I would appreciate a lot if you can give a more detailed description for onnx output.

sandeepnmenon, 2023-08-01:
I see the confusion. I dont think `bevfusion/tools/export.py` works because there are some operations that do not directly export to onnx and TRT (Hence this repo).
I meant you can run export scripts mentioned in this repo.
Please refer the readme for [Quantization for BEVFusion](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/qat/README.md) where they have mentioned the below commands to export the INT model

```
python qat/export-camera.py --ckpt=model/resnet50int8/bevfusion_ptq.pth
python qat/export-transfuser.py --ckpt=model/resnet50int8/bevfusion_ptq.pth
python qat/export-scn.py --ckpt=model/resnet50int8/bevfusion_ptq.pth --save=qat/onnx_int8/lidar.backbone.onnx
```

After you get that, you need to build them using the `build_trt_engine.sh` script to generate the TRT plans.

cdefg, 2023-08-02:
> I see the confusion. I dont think `bevfusion/tools/export.py` works because there are some operations that do not directly export to onnx and TRT (Hence this repo). I meant you can run export [...]

Thank you for your answer!
After a day's work, I can successfully deploy **resnet50 detection model** and test it's mAP. Yet I'm still working more to find 1.how to deploy segmentation or mapping task model; 2.how to deploy swint backbone model. prob(1.) is more important for me, not as (2.).

For (1.) **seg** model I would do some work on model config to change its structure, and change `qat/export-*.py` as they are made natively for detection model. I think it's not hard work.
As for swint(2), I directly used `qat/ptq.py` to export swint config and model (seg and det, both) from bevfusion's original repo and still found problem like original problem in this issue. Now I think that's because this nvidia's repo doesn't mean to support swint model, as the config path just have [configs](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/CUDA-BEVFusion/configs)/[nuscenes](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/CUDA-BEVFusion/configs/nuscenes)/[det](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/CUDA-BEVFusion/configs/nuscenes/det)/[transfusion](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/CUDA-BEVFusion/configs/nuscenes/det/transfusion)/[secfpn](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/CUDA-BEVFusion/configs/nuscenes/det/transfusion/secfpn)/[camera+lidar](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/CUDA-BEVFusion/configs/nuscenes/det/transfusion/secfpn/camera%2Blidar)
/resnet50/convfuser.yaml (not sure).

If you have any idea for my problems, thanks for discussion.

sandeepnmenon, 2023-08-02:
Glad you were able to deploy!
I was able to run the swint backbone model but without quantization ( not running `ptq.py` and just running the exports)

As for the seg model, yes you need to write the hook functions for the segmentation heads and the LSSFPN backbone. But the troubles don't end there, ([issue](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/issues/130)) the `F.grid_sample` function used in `BEVGridTransform` is not supported my any TensorRT ops, so you would need to add a custom plugin to build the TRT plan. After that change the bevfusion.cpp code to load that and run inference.

slayerlpj, 2024-08-02:
> Glad you were able to deploy! I was able to run the swint backbone model but without quantization ( not running `ptq.py` and just running the exports)
>
> As for the seg model, yes you need to write [...]

Hello, How do you run the swint backbone model but without quantization? 
When I used "build_trt_engine.sh" to generate plan file, I got a error as follow:

Failed to build model model/swint/build/head.bbox.plan.
You can check the error message by model/swint/build/head.bbox.log
[08/02/2024-11:42:32] [E] [TRT] ModelImporter.cpp:729: --- End node ---
[08/02/2024-11:42:32] [E] [TRT] ModelImporter.cpp:732: ERROR: builtin_op_importers.cpp:5410 In function importFallbackPluginImporter:
[8] Assertion failed: creator && "Plugin not found, are the plugin name, version, and namespace correct?"
[08/02/2024-11:42:32] [E] Failed to parse onnx file
[08/02/2024-11:42:32] [I] Finish parsing network model
[08/02/2024-11:42:32] [E] Parsing model failed
[08/02/2024-11:42:32] [E] Failed to create engine from model or file.
[08/02/2024-11:42:32] [E] Engine set up failed
&&&& FAILED TensorRT.trtexec [TensorRT v8501] # trtexec --onnx=model/swint/head.bbox.onnx --fp16 --inputIOFormats=fp16:chw, --outputIOFormats=fp16:chw,fp16:chw,fp16:chw,fp16:chw,fp16:chw,fp16:chw, --saveEngine=model/swint/build/head.bbox.plan --memPoolSize=workspace:2048 --verbose --dumpLayerInfo --dumpProfile --separateProfileRun --profilingVerbosity=detailed --exportLayerInfo=model/swint/build/head.bbox.json

jorgemn9, 2024-09-19:
> Glad you were able to deploy! I was able to run the swint backbone model but without quantization ( not running `ptq.py` and just running the exports)
>
> As for the seg model, yes you need to write [...]

Hi @sandeepnmenon, now I have trained BEVFusion in my own dataset with swint and I'm facing the same problem. How did you export the .onnx files from the swint .pth? When I try to export the onnx files without quantization I get the following error "AttributeError: 'dict' object has no attribute 'module' "

Thanks in advance.