NVIDIA-AI-IOT/Lidar_AI_Solution #139: Exporting PTQ model on my custom bevfusion trained weights

sandeepnmenon, 2023-07-26:
Thank you for the amazing work. I was able to setup the BEVFusion inference using the model files given in the readme.
I want to use this pipeline for BEVFusion trained on my dataset, so as per the [Quantization README](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/qat/README.md),
1. Run the export scripts for each module with the checkpoint as my checkpoint that I get from BEVFusion training pipeline
2. Run `qat/ptq.py` with the dataset changed to my custom dataset

TLDR
Can I directly use my BEVFusion weights trained on my dataset in the quantisation flow mentioned in this repo with the change in the `qat/ptq.py` file to use my custom dataset

cdefg, 2023-07-31:
same question, same failure
Need an answer, the same.

sandeepnmenon, 2023-07-31:
Seems like the `qat/ptq.py` is written for the resnet50 backbone. So I think even for custom dataset, as long as the model architecture doesn't change, the script should work as is (except for the change in the dataloaders)