NVIDIA-AI-IOT/Lidar_AI_Solution #142: tensorrt 8.4.1 also works

pegasus-zhang, 2023-07-31:
Hi,
we tried tensorrt 8.4.1 on orin, it also works for "Building the models for tensorRT" and "Compile and run the program",  so  why the Prerequisite is "TensorRT >= 8.5.0" ?

hopef, 2023-08-25:
Thanks for the tip. In our live testing, we have found that versions lower than 8.5.0 may have some issues. 
That's why we gave this recommendation. 
We don't actually have a dependency on the TensorRT version because we don't have a special layer.
Thanks.

paladin1410, 2024-11-30:
@pegasus-zhang Hi, may I ask if you are mention about bev_fusion model as I try to build the project for Orin with  tensorrt 8.4.1 but it does not work due to file src/common/tensorrt.cpp seems to used tensorrt 8.5. Thanks