NVIDIA-AI-IOT/Lidar_AI_Solution #147: multiple output  or different input for libspconv engine

HuangVictorAuto, 2023-08-07:
Hi, 

Does the libspconv engine only support the original 'scn' model that has _voxel,voxel_indics, sparse_shape_ as input and only one DTensor output?

`Exported virtual DTensor* forward(const std::vector<int64_t>& features_shape, DType features_dtype, void* features_data,
                          const std::vector<int64_t>& indices_shape, DType indices_dtype, void* indices_data,
                                      int batch, std::vector<int> grid_size, void* stream = nullptr) = 0;`

what if I want to have different input and multiple output?

Thanks!

HuangVictorAuto, 2023-08-08:
@hopef , thanks for the label as Feature Request, we hope that nvidia can open source the libspconv code, so that we can update the code accordingly by ourself. do you have any plan for the open source ?

hopef, 2023-08-22:
Please refer to [libspconv-1.1.0](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/libraries/3DSparseConvolution/libspconv-1.1.0).

closed 2023-08-22 by hopef