NVIDIA-AI-IOT/Lidar_AI_Solution #149: Replication of BEVFusion PTQ Error

ihaohe, 2023-08-07:
Thanks for your nice work! 
 I've reproduced BEVFusion PTQ performace with the model（Resnet50） you provided and the script **tools/test-mAP-for-cuda.py**   following [https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/qat/README.md](url) to regenerate Renset50-PTQ model.

But when I train my own  BEVFusionModel-Resnet50 (Got 68.10mAP 71.13NDS like yours) and try to use PTQ ，the PTQ process done successfully，however，the nuScenes eval terminate just like:
[image]

I printed the boxes info and found some boxes are "nan"!
[image]

I suspect there is something wrong in the PTQ process，but I've no idea how to debug it，can you give me some suggestions?

liuanqi-libra7, 2023-08-08:
@ihaohe  Could you please provide the version information for your CUDA and TensorRT,?

ihaohe, 2023-08-08:
> @ihaohe Could you please provide the version information for your CUDA and TensorRT?

@liuanqi-libra7 

CUDA 11.3
TensorRT-8.5.1.7
pytorch-quantization  2.1.2
torch  1.10.1+cu113
mmcv 1.4.0
mmdet 2.20.0

My BEVFusion model is here [model](https://drive.google.com/file/d/11b9VI8L6jzUIKS6Ep5NSlpMHzlYRD9n-/view?usp=drive_link) . You can use it to reproduce the error I've met.
Thanks for your help.

ihaohe, 2023-08-08:
My BEVFusion model Int8 result [onnx_int8](https://drive.google.com/file/d/1cb7zDyk512AJbwDRTbc9XI4MXQTfKiMc/view?usp=drive_link). Hope it can help you to locate the problem.

ihaohe, 2023-08-18:
@hopef  @liuanqi-libra7 
I'm sorry to bother you. Is there any progress about this issue？

Anmidy, 2024-12-13:
> Thanks for your nice work! I've reproduced BEVFusion PTQ performace with the model（Resnet50） you provided and the script **tools/test-mAP-for-cuda.py** following [https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/qat/README.md](url) to regenerate Renset50-PTQ model.
> 
> But when I train my own BEVFusionModel-Resnet50 (Got 68.10mAP 71.13NDS like yours) and try to use PTQ ，the PTQ process done successfully，however，the nuScenes eval terminate just like: ![c75550bf1f5942485a20139dfc86dd3a](https://private-user-images.githubusercontent.com/41662686/258765584-0b49ec60-000a-43ee-bcc5-7dfd25bd598e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQwNzg3NzksIm5iZiI6MTczNDA3ODQ3OSwicGF0aCI6Ii80MTY2MjY4Ni8yNTg3NjU1ODQtMGI0OWVjNjAtMDAwYS00M2VlLWJjYzUtN2RmZDI1YmQ1OThlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEzVDA4Mjc1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU5OTBkYjI1ZWE2MzFjNGM1NDhjODRkMmViNTdlNWIyNjc4OTk2Mjg0ZGMzN2VmNWYwMWEyMTYxZWM4ODFkYjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0._g2XUwolowKuMicTOg5fyVPGqInAApsnE9IsA18US7k)
> 
> I printed the boxes info and found some boxes are "nan"! ![image](https://private-user-images.githubusercontent.com/41662686/258765621-4b2134f2-5a8e-4d82-a2eb-eb3154bf9c4f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQwNzg3NzksIm5iZiI6MTczNDA3ODQ3OSwicGF0aCI6Ii80MTY2MjY4Ni8yNTg3NjU2MjEtNGIyMTM0ZjItNWE4ZS00ZDgyLWEyZWItZWIzMTU0YmY5YzRmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEzVDA4Mjc1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE4MDk3Yzc1ZWIxMTVlM2ZlZWIyYmJhNzc3YjI0ZWEzZTMxMmU0MWVhNDdhMTcwMDFjZTExOWIzMTliZTA0MjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.CL5CCDdDDX0WgCpmEZYmYQj-tRMInC5qCBYoAOAyKkA)
> 
> I suspect there is something wrong in the PTQ process，but I've no idea how to debug it，can you give me some suggestions?

你好，请问你训练resnet50为backbone的bevfusion-det.pth模型用的训练工程代码是哪个revision？configs用的是mit的centerhead中的resnet50还是cuda-bevfusion中的configs？