NVIDIA-AI-IOT/Lidar_AI_Solution #150: Can libspconv.so support libprotobuf.so.14

qqcx, 2023-08-08:
Hello. Can it support libprotobuf.so.14? libprotobuf.so.14 is commonly used in programming.

ldd ../3rdparty/libspconv/lib/x86_64/libspconv.so 
	linux-vdso.so.1 (0x00007ffd339cd000)
	libcudart.so.11.0 => /usr/local/cuda/targets/x86_64-linux/lib/libcudart.so.11.0 (0x00007f8f21279000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f8f20ef0000)
	libprotobuf.so.17 => /usr/local/protobuf317/lib/libprotobuf.so.17 (0x00007f8f206aa000)

qqcx, 2023-08-09:
How to make libspconv.so by myself?
I need to use protobuf 3.6.1.3

qqcx, 2023-08-17:
I have solved the problem now.
libprotobuf.so.17 works.

closed 2023-08-17 by qqcx

hcheng1005, 2023-10-26:
> I have solved the problem now. libprotobuf.so.17 works.

how do solved this problem,I meet same problem.Looking forward for your reply,thanks!!!!

hopef, 2023-10-26:
(10/19/2023) The libspconv.so 1.1.1 is released now!
- Fix an issue with wrong indices for intermediate output.

(8/18/2023) The libspconv.so 1.1.0 is released now!
- Open sourced the onnx parser part. libprotobuf will be configured by yourself.
- Multiple outputs are supported.

You can find the new version of libspconv [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/libraries/3DSparseConvolution).

hcheng1005, 2023-10-27:
> (10/19/2023) The libspconv.so 1.1.1 is released now!
>
> * Fix an issue with wrong indices for intermediate output.
>
> (8/18/2023) The libspconv.so 1.1.0 is released now!
>
> * Open sourced the onnx parser [...]

Could it work well on Ubuntu 18.04 version? I do run it well on Ubuntu 20.04 but meet error on Ubuntu18.84

hcheng1005, 2023-10-27:
> (10/19/2023) The libspconv.so 1.1.1 is released now!
>
> * Fix an issue with wrong indices for intermediate output.
>
> (8/18/2023) The libspconv.so 1.1.0 is released now!
>
> * Open sourced the onnx parser [...]

Thanks a lot. I will try later and close question if it works.

hcheng1005, 2023-10-27:
> (10/19/2023) The libspconv.so 1.1.1 is released now!
>
> * Fix an issue with wrong indices for intermediate output.
>
> (8/18/2023) The libspconv.so 1.1.0 is released now!
>
> * Open sourced the onnx parser [...]

It works. Thanks again!