NVIDIA-AI-IOT/Lidar_AI_Solution #153: SparseConvolution

Endlessfancy, 2023-08-15:
Hi! I want to print out the output layer by layer, so I used the Python ONNX library's onnx.checker.check_model function to load the ONNX model of lidarscn. However, I encountered an error:

onnx.onnx_cpp2py_export.checker.ValidationError: No opset import for domain '' ==> Context: Bad node spec for node. Name: conv0 OpType: SparseConvolution

I would like to ask how I can load the SparseConvolution operator into it, or if there's any other suggested way to output the results of each layer?

Thank you!

hopef, 2023-08-16:
The SparseConvolution operator can only be loaded by libspconv.so. 
It is not supported by onnx.checker, onnxruntime and TensorRT.

Endlessfancy, 2023-08-17:
Ok! Thank you for your answer!

Endlessfancy, 2023-08-17:
Also, do you have some other suggestions for printing the output layer by layer?

closed 2023-08-17 by Endlessfancy