NVIDIA-AI-IOT/Lidar_AI_Solution #155: There are any ref to visualize result using python?

SeungHwi0613, 2023-08-16:
[image]

I made it working on ROS2 in realtime. 
I visualize it using my own code, but it seems like having wrong coordinate. 
I want to know, this result is occuured by 'visualization' or 'input' format!
So i need more accurate visualization code like you do using C++

If you give me any advise, it could be so helpful! Thanks.

hopef, 2023-08-16:
It seems the wrong transformation matrix is used.
Here is a recommended video that I hope can help you: https://www.bilibili.com/video/BV1ZX4y187yh

SeungHwi0613, 2023-08-16:
This ROS2 bag file is using nuScenes dataset. (v1-mini scene61) and when i tried this ROS2 migration to the other repo, it was totally correct. So i want to know for future work which scene number did you use for example? I want to compare your tensor input of calibration data with mine.

SeungHwi0613, 2023-08-16:
[image]
it's the result from exactly same transformation matrix

SeungHwi0613, 2023-08-16:
[image]
And it's the result after filtering by score>0.5

SeungHwi0613, 2023-08-16:
[image]
If i modify every z_corner of bounding box to down with some offset(-0.8), it looks like that.

SeungHwi0613, 2023-08-16:
Still do you guess it is the problem of transformation matrix? or just code of visualization?