NVIDIA-AI-IOT/Lidar_AI_Solution #159: 你好，运行bash tool/run.sh报错

xqh5201314, 2023-08-21:
Assert failed 💀. self_bytes == type_bytes in file /home/ni/Lidar_AI_Solution/CUDA-BEVFusion/src/common/tensor.cu:710, message: Failed to check the data type, your code may have a logic error. The type of this tensor is 0 bytes, but the pointer is 4 bytes.
tool/run.sh：行 41: 734893 已放弃               （核心已转储） ./build/bevfusion $DEBUG_DATA $DEBUG_MODEL $DEBUG_PRECISION

closed 2023-08-21 by xqh5201314

sangjinpark97, 2023-09-20:
How to solve it?

J-xinyu, 2023-12-24:
> How to solve it?

do you have any solution？

guangqianzhang, 2024-01-13:
> Assert failed 💀. self_bytes == type_bytes in file /home/ni/Lidar_AI_Solution/CUDA-BEVFusion/src/common/tensor.cu:710, message: Failed to check the data type, your code may have a logic error. The [...]

hello，what is the solution to this problem? would you mind give a answer。thank

ruishanyin, 2024-06-11:
./build/bevfusion $DEBUG_DATA $DEBUG_MODEL $DEBUG_PRECISION,   the $DEBUG_DATA defaut path is example-data in too/environment.sh, make sure  its correct for u.