NVIDIA-AI-IOT/Lidar_AI_Solution #160: this error when I export epoch.pth to onnx with the export_neck_head.py, but the filename.onnx can work, please help me

HULoaf, 2023-08-23:
Warning: Constant folding - Only steps=1 can be constant folded for opset >= 10 onnx::Slice op. Constant folding not applied.
Warning: Constant folding - Only steps=1 can be constant folded for opset >= 10 onnx::Slice op. Constant folding not applied.
[PASS] Export ONNX done.
/data/chuxikai/anaconda3/envs/nvi_cen/lib/python3.8/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  ../aten/src/ATen/native/TensorShape.cpp:2157.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Traceback (most recent call last):
  File "tool/export_neck_head.py", line 299, in <module>
    main(args)
  File "tool/export_neck_head.py", line 280, in main
    np.testing.assert_almost_equal(box3d_lidar, gt_output_ori['box3d_lidar'].cpu().numpy(), decimal=3)
  File "/data/chuxikai/anaconda3/envs/nvi_cen/lib/python3.8/site-packages/numpy/testing/_private/utils.py", line 581, in assert_almost_equal
    return assert_array_almost_equal(actual, desired, decimal, err_msg)
  File "/data/chuxikai/anaconda3/envs/nvi_cen/lib/python3.8/site-packages/numpy/testing/_private/utils.py", line 1044, in assert_array_almost_equal
    assert_array_compare(compare, x, y, err_msg=err_msg, verbose=verbose,
  File "/data/chuxikai/anaconda3/envs/nvi_cen/lib/python3.8/site-packages/numpy/testing/_private/utils.py", line 842, in assert_array_compare
    raise AssertionError(msg)
AssertionError: 
Arrays are not almost equal to 3 decimals

Mismatched elements: 37 / 1683 (2.2%)
Max absolute difference: 38.031326
Max relative difference: 8.296275
 x: array([[ 3.433e-01,  3.039e+01,  5.712e-01, ...,  3.006e-01,  8.440e+00,
        -3.089e+00],
       [-9.198e+00, -1.790e+01, -1.655e+00, ..., -1.204e-01, -4.462e+00,...
 y: array([[ 3.433e-01,  3.039e+01,  5.712e-01, ...,  3.004e-01,  8.440e+00,
        -3.089e+00],
       [-9.198e+00, -1.790e+01, -1.655e+00, ..., -1.203e-01, -4.461e+00,...

hopef, 2023-08-23:
You can comment np.assert_almost_equal function to avoid this error.

HULoaf, 2023-08-23:
> You can comment np.assert_almost_equal function to avoid this error.

Thanks, it works! By the way, i want to train my custom dataset with only lidar, i have tried for long time, but i filed.  Can you give me some advice?  Tears of gratitude

HULoaf, 2023-08-23:
> You can comment np.assert_almost_equal function to avoid this error.

Besides, I have trained  my custom dataset with only lidar on MMdetection3d, but I failed to export onnx from the pth i trained with using your export tools including export-scn.py and export_neck_head.py,  hoping you give me some advice, appreciate!

HULoaf, 2023-08-23:
> You can comment np.assert_almost_equal function to avoid this error.

Besides, I have trained my custom dataset with only lidar on MMdetection3d, but I failed to export onnx from the pth i trained with using your export tools including export-scn.py and export_neck_head.py, could you give me some advice, appreciate!

hopef, 2023-08-25:
I'm pretty sure it's a difficult process, which means you need to learn the exported code and more knowledge of the model. Thanks!