NVIDIA-AI-IOT/Lidar_AI_Solution #161: About camera-geometry.cu

pangxiaowen, 2023-08-23:
When coords.x,coords.y, coords.z are both 0,  Whether keep_count should be incremented。
This happens again when I use my own transformation matrix。

```
    bool kept = coords.x >= 0 && coords.y >= 0 && coords.z >= 0 && coords.x < nx.x && coords.y < nx.y && coords.z < nx.z;
    if (!kept) {
      ranks[_pid] = 0;
    } else {
      atomicAdd(keep_count, 1);
      ranks[_pid] = (coords.x * nx.y + coords.y) * nx.z + coords.z;
    }
```

hopef, 2023-08-23:
Is this wrong?

pangxiaowen, 2023-08-23:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

pangxiaowen, 2023-08-23:
Could cause geometry[indices[interval_starts[i] + remain]] to be a negative number, When coords.x,coords.y, coords.z are both 0, ranks[_pid] = 0.  stable_sort_by_key does not guarantee that all available data are contiguous

CV-player, 2023-12-08:
I am faced with the same problem. Do you fix the bug?

pangxiaowen, 2023-12-11:
> I am faced with the same problem. Have you fixed the bug?
ranks[_pid] = 0;  ----> ranks[_pid] = -1;

    bool kept = coords.x >= 0 && coords.y >= 0 && coords.z >= 0 && coords.x < nx.x && coords.y < nx.y && coords.z < nx.z;
    if (!kept) {
      ranks[_pid] = -1;
    } else {
      atomicAdd(keep_count, 1);
      ranks[_pid] = (coords.x * nx.y + coords.y) * nx.z + coords.z;
    }

hopef, 2023-12-11:
Please give me some time, thansk.