NVIDIA-AI-IOT/Lidar_AI_Solution #164: How to transform nv::Tensor scene_device_image to CV:: Mat format?

Gigalomanicx, 2023-08-28:
in /src/main.cpp   function visualize{ }
```C++
  auto visualizer = nv::create_image_artist(image_artist_param);
  for (size_t icamera = 0; icamera < images.size(); ++icamera) {
    int ox = offset_cameras[icamera][0] + content_width / 2;
    int oy = offset_cameras[icamera][1] + content_height / 2;
    bool xflip = static_cast<bool>(offset_cameras[icamera][2]);
    visualizer->draw_prediction(icamera, predictions, xflip);

    nv::Tensor device_image(std::vector<int>{900, 1600, 3}, nv::DataType::UInt8);
    device_image.copy_from_host(images[icamera], stream);

    if (xflip) {
      auto clone = device_image.clone(stream);
      scene->flipx(clone.ptr<unsigned char>(), clone.size(1), clone.size(1) * 3, clone.size(0), device_image.ptr<unsigned char>(),
                   device_image.size(1) * 3, stream);
      checkRuntime(cudaStreamSynchronize(stream));
    }
    visualizer->apply(device_image.ptr<unsigned char>(), stream);

    scene->resize_to(device_image.ptr<unsigned char>(), ox, oy, ox + camera_width, oy + camera_height, device_image.size(1),
                     device_image.size(1) * 3, device_image.size(0), 0.8f, stream);
    checkRuntime(cudaStreamSynchronize(stream));
  }
  // origin
  // printf("Save to %s\n", save_path.c_str());
  // stbi_write_jpg(save_path.c_str(), scene_device_image.size(1), scene_device_image.size(0), 3,
  //                 scene_device_image.to_host(stream).ptr(), 100);
  
  cv::Mat result_image(scene_device_image.size(0), scene_device_image(1), CV_8UC3);
  auto scene_host_ptr = scene_device_image.to_host(stream).ptr();
  memcpy(result_image.data, scene_host_ptr, result_image.total() * result_image.eleSize());
  return result_image;
```
A core dump occurred, So I think there was an error in my transform. What is the correct transform? And how much time does this conversion take? What is a better way to convert it to a sensor_ Msgs/Image for ROS?

closed 2023-08-28 by Gigalomanicx

pangxiaowen, 2023-09-04:
I had a similar problem, how did you solve it。

Gigalomanicx, 2023-09-05:
@pangxiaowen in /src/main.cpp function visualize{ }
```C++

    if (xflip) {
      auto clone = device_image.clone(stream);
      scene->flipx(clone.ptr<unsigned char>(), clone.size(1), clone.size(1) * 3, clone.size(0), device_image.ptr<unsigned char>(),
                   device_image.size(1) * 3, stream);
      checkRuntime(cudaStreamSynchronize(stream));
    }
    visualizer->apply(device_image.ptr<unsigned char>(), stream);

    scene->resize_to(device_image.ptr<unsigned char>(), ox, oy, ox + camera_width, oy + camera_height, device_image.size(1),
                     device_image.size(1) * 3, device_image.size(0), 0.8f, stream);
    checkRuntime(cudaStreamSynchronize(stream));
  }
  // 创建cv::Mat空间
  cv::Mat result_image(scene_device_image.size(0), scene_device_image.size(1), CV_8UC3);
  // GPU -> CPU
  // memory copy [dest, src, size]
  memcpy(result_image.data, scene_device_image.to_host(stream).ptr() , result_image.total() * result_image.elemSize());
  return result_image;
}
```
in /src/main.cpp function sync{ }
```C++
//   // 加载图像和点云数据
  std::vector<unsigned char*> images;
  std::vector<sensor_msgs::Image::ConstPtr> image_msgs = {f_msg,fr_msg,fl_msg,b_msg,bl_msg,br_msg};
  for(const auto& image_msg : image_msgs) {
    cv_bridge::CvImagePtr cv_ptr;
    try{
      cv_ptr = cv_bridge::toCvCopy(image_msg, sensor_msgs::image_encodings::BGR8);
      int img_length = cv_ptr->image.total() * cv_ptr->image.channels();
      unsigned char* image = new unsigned char[img_length]();
      std::memcpy(image, cv_ptr->image.ptr<unsigned char>(0), img_length * sizeof(unsigned char));
      images.push_back(image);
    }
    catch (cv_bridge::Exception& e){
        ROS_ERROR("cv_bridge exception: %s", e.what());
        return;
    }
  }
  auto lidar_points = cloud_cb(lt_msg);
  // warmup
  auto bboxes =
      core->forward((const unsigned char**)images.data(), lidar_points.ptr<nvtype::half>(), lidar_points.size(0), stream);

  // 输出
  // cv::Mat格式 结果图像
  cv::Mat processed_image = visualize(bboxes, lidar_points, images, lidar2image, stream);

  // destroy memory
  for (size_t i = 0; i < images.size(); ++i) stbi_image_free(images[i]);
  images.clear();

  // 结果图像处理完成，将处理后的图像转换为ROS sensor_msgs格式发布
  cv_bridge::CvImage cv_image;
  cv_image.image = processed_image;
  cv_image.encoding = "bgr8";
  sensor_msgs::ImagePtr processed_msg = cv_image.toImageMsg();

  // 发布
  image_pub.publish(processed_msg);
}
```

leslie27ch, 2023-09-08:
> ```c++
> new unsigned char[img_length]();
> ```

@Gigalomanicx  hi，I use your code and run it， error is： free（）：invalid pointer aborted （core dumped）。
problem 1： new char[]  should use delete[] ，not use free
[image]

when i change new char to malloc， the error is still。

do you solve the problem ？

another problem ：
and when i use ros receive rosmsg of Images ， I move code to  msg callback function from main.cpp。  after server time，the cuda is out of memory：
[image]
[image]
[image]

thankyou very much

Gigalomanicx, 2023-09-09:
my main func is here，I use message_filters::Synchronizer to sync different topic msg，then i use ts_callback to run it

for problem 1&2：
I don't have the problem you're talking about, but I think maybe there's something wrong with the way your topic msg calleback.
and I think ‘cudaStreamCreate(&stream);’ should not be created more than once in a callback.

With best wishes,
Gigalomanicx.