NVIDIA-AI-IOT/Lidar_AI_Solution #166: About voxelization parameters

catziyan, 2023-09-03:
The voxelization parameters I used during training are
  min_range: [-27.0, 0, -5.0]
  max_range: [27.0, 54.0, +3.0]
  voxel_size: [0.075, 0.075, 0.2]

and I successfully converted the trained model to ONNX. However, when I modified the voxelization parameter in the code according to the parameters used during training, I got incorrect inference results. Surprisingly, not modifying the parameters yielded better results. Why is this?

hopef, 2023-09-04:
Maybe you should double-check your training script to see what has happened.

qyang1996, 2023-09-06:
I am also meet this problem. When I use default parameters, I can obtain the correct result, but when I expand the "max_x_range", I also got wrong results.  
I check the input of scn_engine. Only sparse_shape=[41, 1440, 1440] can got the correct result.

hopef, 2023-09-06:
I'm sorry. I'll check it soon.

Could you provide the wrong code and model?

eugenebak, 2024-05-17:
Is there any update about this issue?

ruishanyin, 2024-07-02:
> I am using the CenterPoint model trained on my own data. training config is: ` voxel_generator = dict( range=[-30.0, -75.0, -5.0, 120.0, 75.0, 3.0], voxel_size=[0.075, 0.075, 0.2] )` The sparse shape will be[40, 2000, 2000].
> 
> When I modify these parameters in the common.h file, I got the incorrect 3D box. I compare the python reuslt of generateVoxels with this project, it is similar. When I manually update the sparse shape to [41, 1440, 1440] befor `scn_engine_->forward()`, I can get correct 3D box.

Did u transfer your data to nuscenes ? How?

ruishanyin, 2024-07-02:
@qyang1996