NVIDIA-AI-IOT/Lidar_AI_Solution #168: Inference time Jetson agx orin developer kit

dav695, 2023-09-06:
Hello, 
I am very interested in using this repository for my project, but when I follow the execution steps with the example data I get the following inference times: 

```
root@ubuntu:/home/Lidar_AI_Solution/CUDA-BEVFusion# bash tool/run.sh
==========================================================
||  MODEL: resnet50int8
||  PRECISION: int8
||  DATA: example-data
||  USEPython: ON
||
||  TensorRT: /usr/src/tensorrt/lib
||  CUDA: /usr/local/cuda
||  CUDNN: /usr/lib/aarch64-linux-gnu/libcudnn.so
==========================================================
Find Python_Inc: /usr/include/python3.8
Find Python_Lib: /usr/lib
Find Python_Soname: [libpython3.8.so](http://libpython3.8.so/)
Try to get the current device SM
Current CUDA SM: 87
Configuration done!
-- Configuring done
-- Generating done
-- Build files have been written to: /home/Lidar_AI_Solution/CUDA-BEVFusion/build
Consolidate compiler generated dependencies of target bevfusion_core
[ 58%] Built target bevfusion_core
Consolidate compiler generated dependencies of target pybev
Consolidate compiler generated dependencies of target bevfusion
[ 66%] Built target pybev
[100%] Built target bevfusion
Create by resnet50int8, int8
------------------------------------------------------
Camerea Backbone 🌱 is Static Shape model
Inputs: 2
0.img : {1 x 6 x 3 x 256 x 704} [Float16]
1.depth : {1 x 6 x 1 x 256 x 704} [Float16]
Outputs: 2
0.camera_depth_weights : {6 x 118 x 32 x 88} [Float16]
1.camera_feature : {6 x 32 x 88 x 80} [Float16]
------------------------------------------------------
------------------------------------------------------
Camerea VTransform 🌱 is Static Shape model
Inputs: 1
0.feat_in : {1 x 80 x 360 x 360} [Float16]
Outputs: 1
0.feat_out : {1 x 80 x 180 x 180} [Float16]
------------------------------------------------------
------------------------------------------------------
Transfusion 🌱 is Static Shape model
Inputs: 2
0.camera : {1 x 80 x 180 x 180} [Float16]
1.lidar : {1 x 256 x 180 x 180} [Float16]
Outputs: 1
0.middle : {1 x 512 x 180 x 180} [Float16]
------------------------------------------------------
------------------------------------------------------
BBox 🌱 is Static Shape model
Inputs: 1
0.middle : {1 x 512 x 180 x 180} [Float16]
Outputs: 6
0.reg : {1 x 2 x 200} [Float16]
1.height : {1 x 1 x 200} [Float16]
2.dim : {1 x 3 x 200} [Float16]
3.rot : {1 x 2 x 200} [Float16]
4.vel : {1 x 2 x 200} [Float16]
5.score : {1 x 10 x 200} [Float16]
------------------------------------------------------
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 1.76112 ms
[⏰ [NoSt] ImageNrom]: 8.97926 ms
[⏰ Lidar Backbone]: 70.71379 ms
[⏰ Camera Depth]: 0.36797 ms
[⏰ Camera Backbone]: 53.43235 ms
[⏰ Camera Bevpool]: 5.97578 ms
[⏰ VTransform]: 12.66403 ms
[⏰ Transfusion]: 34.16090 ms
[⏰ Head BoundingBox]: 42.21536 ms
Total: 219.530 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 1.07373 ms
[⏰ [NoSt] ImageNrom]: 9.14886 ms
[⏰ Lidar Backbone]: 62.72061 ms
[⏰ Camera Depth]: 0.36950 ms
[⏰ Camera Backbone]: 44.97417 ms
[⏰ Camera Bevpool]: 5.83846 ms
[⏰ VTransform]: 9.81789 ms
[⏰ Transfusion]: 30.96118 ms
[⏰ Head BoundingBox]: 43.23085 ms
Total: 197.913 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 1.05382 ms
[⏰ [NoSt] ImageNrom]: 8.28922 ms
[⏰ Lidar Backbone]: 54.39805 ms
[⏰ Camera Depth]: 0.35386 ms
[⏰ Camera Backbone]: 48.16608 ms
[⏰ Camera Bevpool]: 7.36582 ms
[⏰ VTransform]: 10.33014 ms
[⏰ Transfusion]: 30.05856 ms
[⏰ Head BoundingBox]: 39.00249 ms
Total: 189.675 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 0.60365 ms
[⏰ [NoSt] ImageNrom]: 5.60826 ms
[⏰ Lidar Backbone]: 56.04077 ms
[⏰ Camera Depth]: 0.41162 ms
[⏰ Camera Backbone]: 49.25574 ms
[⏰ Camera Bevpool]: 5.88304 ms
[⏰ VTransform]: 11.25626 ms
[⏰ Transfusion]: 33.49014 ms
[⏰ Head BoundingBox]: 40.32442 ms
Total: 196.662 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 1.11021 ms
[⏰ [NoSt] ImageNrom]: 8.24848 ms
[⏰ Lidar Backbone]: 63.06800 ms
[⏰ Camera Depth]: 0.35357 ms
[⏰ Camera Backbone]: 48.95136 ms
[⏰ Camera Bevpool]: 5.91782 ms
[⏰ VTransform]: 11.86054 ms
[⏰ Transfusion]: 34.76771 ms
[⏰ Head BoundingBox]: 39.79075 ms
Total: 204.710 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 0.59222 ms
[⏰ [NoSt] ImageNrom]: 5.40960 ms
[⏰ Lidar Backbone]: 68.19264 ms
[⏰ Camera Depth]: 0.36458 ms
[⏰ Camera Backbone]: 49.07693 ms
[⏰ Camera Bevpool]: 7.39024 ms
[⏰ VTransform]: 12.18131 ms
[⏰ Transfusion]: 34.83331 ms
[⏰ Head BoundingBox]: 39.59850 ms
Total: 211.638 ms
=============================================
Save to build/cuda-bevfusion.jpg
```
And the output seems to be good. 
[image]

What I am doing wrong?, I think my settings are correct on the Jetson. 
Versions installed:
-Jetpack: 5.2.1
- L4T: 35.4.1
- CUDA: 11.4.315
- cuDNN: 8.6.0.166
- TensorRT: 8.5.2.2

What do I have to do to get this 25 FPS on the platform?

Thank you so much! And I hope you answer me soon.

hopef, 2023-09-06:
You should set the environment by `sudo jetson_clocks` to enable the max freq.

Or, you can try the other command: `sudo nvpmodel -m 0`.

dav695, 2023-09-06:
With the second command and after rebooting the device it seems that it works.
```
root@ubuntu:/home/Lidar_AI_Solution/CUDA-BEVFusion# bash tool/run.sh 
==========================================================
||  MODEL: resnet50int8
||  PRECISION: int8
||  DATA: example-data
||  USEPython: ON
||
||  TensorRT: /usr/src/tensorrt/lib
||  CUDA: /usr/local/cuda
||  CUDNN: /usr/lib/aarch64-linux-gnu/libcudnn.so
==========================================================
Find Python_Inc: /usr/include/python3.8
Find Python_Lib: /usr/lib
Find Python_Soname: [libpython3.8.so](http://libpython3.8.so/)
Try to get the current device SM
Current CUDA SM: 87
Configuration done!
-- Configuring done
-- Generating done
-- Build files have been written to: /home/Lidar_AI_Solution/CUDA-BEVFusion/build
Consolidate compiler generated dependencies of target bevfusion_core
[ 58%] Built target bevfusion_core
Consolidate compiler generated dependencies of target pybev
Consolidate compiler generated dependencies of target bevfusion
[ 66%] Built target pybev
[100%] Built target bevfusion
Create by resnet50int8, int8
------------------------------------------------------
Camerea Backbone 🌱 is Static Shape model
Inputs: 2
0.img : {1 x 6 x 3 x 256 x 704} [Float16]
1.depth : {1 x 6 x 1 x 256 x 704} [Float16]
Outputs: 2
0.camera_depth_weights : {6 x 118 x 32 x 88} [Float16]
1.camera_feature : {6 x 32 x 88 x 80} [Float16]
------------------------------------------------------
------------------------------------------------------
Camerea VTransform 🌱 is Static Shape model
Inputs: 1
0.feat_in : {1 x 80 x 360 x 360} [Float16]
Outputs: 1
0.feat_out : {1 x 80 x 180 x 180} [Float16]
------------------------------------------------------
------------------------------------------------------
Transfusion 🌱 is Static Shape model
Inputs: 2
0.camera : {1 x 80 x 180 x 180} [Float16]
1.lidar : {1 x 256 x 180 x 180} [Float16]
Outputs: 1
0.middle : {1 x 512 x 180 x 180} [Float16]
------------------------------------------------------
------------------------------------------------------
BBox 🌱 is Static Shape model
Inputs: 1
0.middle : {1 x 512 x 180 x 180} [Float16]
Outputs: 6
0.reg : {1 x 2 x 200} [Float16]
1.height : {1 x 1 x 200} [Float16]
2.dim : {1 x 3 x 200} [Float16]
3.rot : {1 x 2 x 200} [Float16]
4.vel : {1 x 2 x 200} [Float16]
5.score : {1 x 10 x 200} [Float16]
------------------------------------------------------
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 1.94266 ms
[⏰ [NoSt] ImageNrom]: 9.31830 ms
[⏰ Lidar Backbone]: 52.33523 ms
[⏰ Camera Depth]: 0.22630 ms
[⏰ Camera Backbone]: 31.26262 ms
[⏰ Camera Bevpool]: 2.32909 ms
[⏰ VTransform]: 2.63872 ms
[⏰ Transfusion]: 13.21322 ms
[⏰ Head BoundingBox]: 20.22544 ms
Total: 122.231 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 0.50422 ms
[⏰ [NoSt] ImageNrom]: 4.45901 ms
[⏰ Lidar Backbone]: 14.63066 ms
[⏰ Camera Depth]: 0.10467 ms
[⏰ Camera Backbone]: 12.74874 ms
[⏰ Camera Bevpool]: 1.50944 ms
[⏰ VTransform]: 1.68672 ms
[⏰ Transfusion]: 8.33939 ms
[⏰ Head BoundingBox]: 14.25162 ms
Total: 53.271 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 0.68451 ms
[⏰ [NoSt] ImageNrom]: 5.19776 ms
[⏰ Lidar Backbone]: 12.21533 ms
[⏰ Camera Depth]: 0.10746 ms
[⏰ Camera Backbone]: 13.50934 ms
[⏰ Camera Bevpool]: 1.48294 ms
[⏰ VTransform]: 1.69533 ms
[⏰ Transfusion]: 7.17760 ms
[⏰ Head BoundingBox]: 13.05619 ms
Total: 49.244 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 0.37190 ms
[⏰ [NoSt] ImageNrom]: 3.39440 ms
[⏰ Lidar Backbone]: 13.26925 ms
[⏰ Camera Depth]: 0.10531 ms
[⏰ Camera Backbone]: 11.82918 ms
[⏰ Camera Bevpool]: 1.48253 ms
[⏰ VTransform]: 1.63264 ms
[⏰ Transfusion]: 7.11702 ms
[⏰ Head BoundingBox]: 11.55440 ms
Total: 46.990 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 0.67645 ms
[⏰ [NoSt] ImageNrom]: 5.54550 ms
[⏰ Lidar Backbone]: 14.05990 ms
[⏰ Camera Depth]: 0.10720 ms
[⏰ Camera Backbone]: 11.84576 ms
[⏰ Camera Bevpool]: 1.47610 ms
[⏰ VTransform]: 1.66339 ms
[⏰ Transfusion]: 7.12205 ms
[⏰ Head BoundingBox]: 12.09120 ms
Total: 48.366 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 0.68240 ms
[⏰ [NoSt] ImageNrom]: 5.17571 ms
[⏰ Lidar Backbone]: 13.37251 ms
[⏰ Camera Depth]: 0.10509 ms
[⏰ Camera Backbone]: 12.22378 ms
[⏰ Camera Bevpool]: 1.49827 ms
[⏰ VTransform]: 1.63914 ms
[⏰ Transfusion]: 9.94246 ms
[⏰ Head BoundingBox]: 11.51382 ms
Total: 50.295 ms
=============================================
Save to build/cuda-bevfusion.jpg
```
Thanks you so much!

closed 2023-09-06 by dav695