NVIDIA-AI-IOT/Lidar_AI_Solution #169: cuOSD Error ，unsupport text backend：StbTrueType

leslie27ch, 2023-09-06:
[image]
[image]

thank you ~

hopef, 2023-09-06:
https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/ae4929b194de662318eb665f95fef91a591c6365/libraries/cuOSD/Makefile#L27C4-L27C4

WangYichen6120, 2025-06-20:
Could you please tell me how to solve this problem, I'm running into the same problem, which causes the detection box in the image to be missing confidence values.

[image]
[image]