NVIDIA-AI-IOT/Lidar_AI_Solution #170: Out of Memory load_engine_from_onnx

NoahLinckeKodiak, 2023-09-07:
Hi, using the new libspconv-1.1.0 I am running into `LLVM ERROR: out of memory` during `load_engine_from_onnx`.

This happens on the the last SparseConvolution layer with rulebook: spconv_down2, and this same error does not happen on the same hardware using the old libspconv.so.

Please let me know if you have any advice, here is the ONNX file for reference: https://drive.google.com/file/d/19mXR-Zvh2pyFRpiuwnm8LHP-lq_cRadc/view?usp=sharing

hopef, 2023-09-07:
Sorry, I can't reproduce this error. Could you provide the full code for this issue?

NoahLinckeKodiak, 2023-09-07:
Yes, here is the minimal example I have been trying:

```
#include "src/onnx-parser.hpp"

int main()
{
    std::string model_path ="/home/n.lincke/Downloads/lidar.backbone.xyz.onnx";
    spconv::Precision precision = spconv::Precision::Float16;
    spconv::load_engine_from_onnx(model_path, precision);
    return 0;
}
```

Where "src/onnx-parser.hpp" is the one provided in `Lidar_AI_Solution/libraries/3DSparseConvolution/src`

The program is crashing in layer 36, during `auto n = builder->push_sparse_conv(...)`

NoahLinckeKodiak, 2023-09-07:
The exact error causing OOM is:

libc++abi: terminating with uncaught exception of type std::bad_alloc: std::bad_alloc

NoahLinckeKodiak, 2023-09-07:
@hopef Would it be possible to upload a second version of libspconv.so that is compiled with Clang's libc++.so instead of GCC libstdc++.so?

I believe this is the problem, as my binary is being compiled with libc++, and then linked to the libstdc++ library. Everything compiles and links since all functions directly called from the binary (load_engine_from_onnx) are being recompiled with libc++. However further internal functions will be called at runtime with the expectation that they are libc++, but may perform other operations according to the libstdc++ that it was compiled with and cause OOM.

hopef, 2023-09-08:
OK, will build a new binary with libc++.

NoahLinckeKodiak, 2023-09-21:
@hopef Hey just wanted to check if you had any updates? Thanks!

hopef, 2023-09-26:
I'm sorry for not getting back to you sooner.
I will update it as soon as possible. Thanks.