NVIDIA-AI-IOT/Lidar_AI_Solution #174: error when replace load_images of loading ros msg

leslie27ch, 2023-09-08:
[image]
hello author， when I  use this loading ros message of 6 images to replace orin load_images，and keep other code is the same。
i meet a error：
[image]

I check load_images， it alse use malloc to store unsigned char image data，and my result.jpg is correct。 only I donot know why the error： double free or corruption （out）

hopef, 2023-09-08:
How did you free the images pointer.

leslie27ch, 2023-09-08:
[image]

and  also try

[image]

leslie27ch, 2023-09-08:
> How did you free the images pointer.

[image]

also try：
[image]

leslie27ch, 2023-09-08:
all above code is in ros message callback function, when I add ros::spin() in the end of message callback function, it will be correctly running, which is very weird.
[image]

Gigalomanicx, 2023-09-09:
@leslie27ch I replied to your email and may be able to answer your question.

rayxie11, 2024-03-12:
Is anyone in this convo generous enough to share the resolution? Many thanks in advance!