NVIDIA-AI-IOT/Lidar_AI_Solution #176: No module named 'det3d'

WYYAHYT, 2023-09-09:
I tried to run `python tool/eval_nusc.py --dump`, got errors like this:
```
Traceback (most recent call last):
  File "tool/eval_nusc.py", line 31, in <module>
    from det3d.torchie import Config
ModuleNotFoundError: No module named 'det3d'
```
I guess there must be a directory named "CenterPoint" in CUDA-CenterPoint, But I can't get it, is there any problem of excuting command `git clone --recursive https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution.git`? But CUDA-CenterPoint seems not a submodule? 

And I trid to clone again, all modules in my workspace are:
[image]

WYYAHYT, 2023-09-09:
Maybe I should clone CenterPoint repo in CUDA-CenterPoint?

hopef, 2023-09-11:
yes

closed 2023-09-12 by WYYAHYT