NVIDIA-AI-IOT/Lidar_AI_Solution #177: Train and test on custom data.

krishnatoshniwal, 2023-09-11:
I want to test the Centerpoint model on some point cloud data that I have. Here's an outline of the steps that I'm currently thinking is required.

1. Train the model with the model definition given in https://github.com/tianweiy/CenterPoint/blob/master/configs/nusc/voxelnet/nusc_centerpoint_voxelnet_0075voxel_fix_bn_z.py
2. Export the model to onnx using - `export-scn.py` and `export_neck_head.py`.

Is that right? Also, what about models trained using mmdet3d, can the sparse conv present in these models also be converted to onnx using `export-scn` ?

hopef, 2023-09-26:
yep, mmdet3d also supported by [export-scn](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/qat/export-scn.py).

ruishanyin, 2024-07-02:
Did u transfer your custom data to nuscenes? How?