NVIDIA-AI-IOT/Lidar_AI_Solution #181: CHECK failed: (index) < (current_size_)

sangjinpark97, 2023-09-18:
Hi,

When I run `bash tool/run.sh`, the below error occurs

```shell
==========================================================
||  MODEL: resnet50
||  PRECISION: int8
||  DATA: custom-data
||  USEPython: ON
||
||  TensorRT: /usr/lib/aarch64-linux-gnu
||  CUDA: /usr/local/cuda-11.4
||  CUDNN: /usr/local/cuda-11.4/lib64
==========================================================
Find Python_Inc: /usr/include/python3.8
Find Python_Lib: /usr/lib
Find Python_Soname: libpython3.8.so
Try to get the current device SM
Current CUDA SM: 87
Configuration done!
CMake Deprecation Warning at CMakeLists.txt:22 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Warning (dev) at CMakeLists.txt:30 (find_package):
  Policy CMP0146 is not set: The FindCUDA module is removed.  Run "cmake
  --help-policy CMP0146" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /home/Lidar_AI_Solution/CUDA-BEVFusion/build
[  4%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_head-transbbox.cu.o
[  8%] Linking CXX shared library libbevfusion_core.so
[ 58%] Built target bevfusion_core
[ 62%] Linking CXX shared library libpybev.so
[ 66%] Linking CXX executable bevfusion
[ 70%] Built target pybev
[100%] Built target bevfusion
Create by resnet50, int8
[libprotobuf FATAL /usr/include/google/protobuf/repeated_field.h:1506] CHECK failed: (index) < (current_size_): 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: (index) < (current_size_): 
tool/run.sh: line 41: 106271 Aborted                 (core dumped) ./build/bevfusion $DEBUG_DATA $DEBUG_MODEL $DEBUG_PRECISION
```

Can you assume the reason?

Thank you in advance!

sangjinpark97, 2023-09-19:
I run `bash tool/run.sh` using model without PTQ.

I get the PTQ model using `qat/ptq.py` and convert it to INT8 model.

But, another error occurs when running `tool/build_trt_engine.sh`

I've uploaded new issue about it in here.

YueSun0609, 2023-09-19:
> I run `bash tool/run.sh` using model without PTQ.
>
> I get the PTQ model using `qat/ptq.py` and convert it to INT8 model.
>
> But, another error occurs when running `tool/build_trt_engine.sh`
>
> I've [...]

I also met the same issue using model without PTQ (run ptq.py but comment the line "model = quantize_net(model)"), maybe that's the reason?

sangjinpark97, 2023-09-19:
@YueSun0609 Did you meet the `CHECK failed` error? 

I directly exported the onnx without `ptq.py` so I'm not sure if the commented line caused the error.

YueSun0609, 2023-09-19:
> @YueSun0609 Did you meet the `CHECK failed` error?
> 
> I directly exported the onnx without `ptq.py` so I'm not sure if the commented line caused the error.
> 
> From now on, I found that the issue arises when the `lidar-scn` model is loaded. But, I can't solve the error.

Yes, I did meet the CHECK failed error. But if directly run export-xx without running ptq.py first, I will met the error that says “attributeerror: 'dict' object has no attribute 'module'”

sangjinpark97, 2023-09-19:
> Yes, I did meet the CHECK failed error. But if directly run export-xx without running ptq.py first, I will met the error that says “attributeerror: 'dict' object has no attribute 'module'”

Then, you need to modify some part of the `export-xx.py`.

One of example is to remove `.module` in [this line](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/ae4929b194de662318eb665f95fef91a591c6365/CUDA-BEVFusion/qat/export-camera.py#L85).

YueSun0609, 2023-09-19:
> Then, you need to modify some part of the `export-xx.py`.
>
> One of example is to remove `.module` in [this [...]

Okay, thank you so much! I think runnung ptq.py first can avoid this issue. I comment some lines in camera related quatization and it works (in my case, data is fp16 and I convert models in fp16). 

```
def quantize_encoders_camera_branch(model_camera_branch):
    quantize_camera_backbone(model_camera_branch.backbone)  
    quantize_camera_neck(model_camera_branch.neck)
    quantize_camera_vtransform(model_camera_branch.vtransform)
    
    '''
    Make all inputs of each concat have the same scale
    Improved performance when using TensorRT forward
    '''

    # major = model_camera_branch.backbone.layer3[0].conv1._input_quantizer
    # model_camera_branch.neck.quant_concat1._input_quantizer = major
    # model_camera_branch.neck.lateral_convs[0].conv._input_quantizer = major
    # model_camera_branch.backbone.layer3[0].downsample[0]._input_quantizer = major
    
    # major = model_camera_branch.backbone.layer4[0].conv1._input_quantizer
    # model_camera_branch.neck.quant_concat0._input_quantizer = major
    # model_camera_branch.neck.lateral_convs[1].conv._input_quantizer = major
    # model_camera_branch.backbone.layer4[0].downsample[0]._input_quantizer = major
```

sangjinpark97, 2023-09-19:
@YueSun0609 Thank you for your suggestion! I'll try it

By the way, what does "data is fp16" mean?
It can be a trivial question. I'm not very familiar with TenorRT. Thank you!

sangjinpark97, 2023-09-20:
@YueSun0609

Hi, I tried your solution and the error is resolved.
But, new error raises when running `bash tool/build_trt_engine.sh` [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/issues/182#issue-1902003328).

If you have any experience about it, let me know how to deal with. Thanks!

sangjinpark97, 2023-09-21:
The error posted in this issue is solved. Closed

closed 2023-09-21 by sangjinpark97

YueSun0609, 2023-09-27:
> @YueSun0609 Thank you for your suggestion! I'll try it
> 
> By the way, what does "data is fp16" mean? It can be a trivial question. I'm not very familiar with TenorRT. Thank you!

This repo provides two precisions, int8 and fp16 (float 16), and I choosed the fp16.

YueSun0609, 2023-09-27:
> I run `bash tool/run.sh` using model without PTQ.
>
> I get the PTQ model using `qat/ptq.py` and convert it to INT8 model.
>
> But, another error occurs when running `tool/build_trt_engine.sh`
>
> I've [...]

Did you also changed quantize.py? I met this

`Traceback (most recent call last):
  File "qat/export-camera.py", line 137, in <module>
    main()
  File "qat/export-camera.py", line 88, in main
    quantize.disable_quantization(model).apply()
  File "/mnt/d373e829-7da0-4037-8630-028a7f795c19/Lidar_AI_Solution/cuda-bevfusion-our-test/qat/lean/quantize.py", line 391, in apply
    for name, module in self.model.named_modules():
AttributeError: 'dict' object has no attribute 'named_modules'
`

sangjinpark97, 2023-09-27:
@YueSun0609 The model seems to be loaded with dictionary, not torch module.

YueSun0609, 2023-09-27:
> @YueSun0609 The model seems to be loaded with dictionary, not torch module.

Yes, this is the bevfusion model I trained use my own data, but I remember if I use the model they provided, I met the same issue. Did you met this?

sangjinpark97, 2023-09-27:
No, I have not met this error.

1. What's the weights you use (let me know the filename)

2. Can you provide the code line where the model is loaded?

sangjinpark97, 2023-09-27:
the checkpoint used in export-camera.py should be PTQ model or .pth which contains torch module, not dict.

bevfusion-det.pth that the author provided seems to be dict about the model parameters.

YueSun0609, 2023-09-27:
> the checkpoint used in export-camera.py should be PTQ model or .pth which contains torch module, not dict.
> 
> bevfusion-det.pth that the author provided seems to be dict about the model parameters.

Yes, but you said "I directly exported the onnx without ptq.py so I'm not sure if the commented line caused the error", so you directly use their _ptq.pth? But if skip ptq.py I can't generate a _ptq.pth model.

sangjinpark97, 2023-09-27:
@YueSun0609 Oh sorry.

When I use the model which is not applied PTQ, I changed the code line where the model loaded.

For example, 

```python
model = BEVFusion() # you should import it
model.load_state_dict(weights.pth)

```

YueSun0609, 2023-09-27:
> @YueSun0609 Oh sorry.
> 
> When I use the model which is not applied PTQ, I changed the code line where the model loaded.
> 
> For example,
> 
> ```python
> model = BEVFusion() # you should import it
> model.load_state_dict(weights.pth)
> ```

Okay, let me try this. Thank you so much!

YueSun0609, 2023-09-27:
> @YueSun0609 Oh sorry.
> 
> When I use the model which is not applied PTQ, I changed the code line where the model loaded.
> 
> For example,
> 
> ```python
> model = BEVFusion() # you should import it
> model.load_state_dict(weights.pth)
> ```

After skipping ptq.py, I met this (index) < (current_size_) error.

YueSun0609, 2023-10-13:
> I run `bash tool/run.sh` using model without PTQ.
>
> I get the PTQ model using `qat/ptq.py` and convert it to INT8 model.
>
> But, another error occurs when running `tool/build_trt_engine.sh`
>
> I've [...]

Hi, may I ask if you can successfully load and use the generated onnx and plan models without PTQ? I can skip PTQ and generate the corresponding onnx and plan files, but if I want to use them by " core = libpybev.load_bevfusion", it will meet this [CHECK failed: (index) < (current_size_)] error. Thank you so much! Looking forward to your reply!

sangjinpark97, 2023-10-16:
@YueSun0609, I could not load the model without PTQ. Now, I'm using the onnx and plan model built with ptq model.