NVIDIA-AI-IOT/Lidar_AI_Solution #182: Assertion failed: scaleAllPositive && "Scale coefficients must all be positive" when build_trt_engine.sh

sangjinpark97, 2023-09-19:
Hi,

I get the PTQ model and convert it to INT8 model.

but, when running `bash tool/build_trt_engine.sh`, building `camera.backbone.plan` is failed.
I can see the below log in `model/resnet50int8/build/camera.backone.log`.

```shell
&&&& RUNNING TensorRT.trtexec [TensorRT v8502] # trtexec --onnx=model/resnet50int8/camera.backbone.onnx --fp16 --int8 --inputIOFormats=fp16:chw,fp16:chw, --outputIOFormats=fp16:chw,fp16:chw, --saveEngine=model/resnet50int8/build/camera.backbone.plan --memPoolSize=workspace:2048 --verbose --dumpLayerInfo --dumpProfile --separateProfileRun --profilingVerbosity=detailed --exportLayerInfo=model/resnet50int8/build/camera.backbone.json
[09/19/2023-09:36:23] [I] === Model Options ===
[09/19/2023-09:36:23] [I] Format: ONNX
[09/19/2023-09:36:23] [I] Model: model/resnet50int8/camera.backbone.onnx
[09/19/2023-09:36:23] [I] Output:
[09/19/2023-09:36:23] [I] === Build Options ===
[09/19/2023-09:36:23] [I] Max batch: explicit batch
[09/19/2023-09:36:23] [I] Memory Pools: workspace: 2048 MiB, dlaSRAM: default, dlaLocalDRAM: default, dlaGlobalDRAM: default
[09/19/2023-09:36:23] [I] minTiming: 1
[09/19/2023-09:36:23] [I] avgTiming: 8
[09/19/2023-09:36:23] [I] Precision: FP32+FP16+INT8
[09/19/2023-09:36:23] [I] LayerPrecisions: 
[09/19/2023-09:36:23] [I] Calibration: Dynamic
[09/19/2023-09:36:23] [I] Refit: Disabled
[09/19/2023-09:36:23] [I] Sparsity: Disabled
[09/19/2023-09:36:23] [I] Safe mode: Disabled
[09/19/2023-09:36:23] [I] DirectIO mode: Disabled
[09/19/2023-09:36:23] [I] Restricted mode: Disabled
[09/19/2023-09:36:23] [I] Build only: Disabled
[09/19/2023-09:36:23] [I] Save engine: model/resnet50int8/build/camera.backbone.plan
[09/19/2023-09:36:23] [I] Load engine: 
[09/19/2023-09:36:23] [I] Profiling verbosity: 2
[09/19/2023-09:36:23] [I] Tactic sources: Using default tactic sources
[09/19/2023-09:36:23] [I] timingCacheMode: local
[09/19/2023-09:36:23] [I] timingCacheFile: 
[09/19/2023-09:36:23] [I] Heuristic: Disabled
[09/19/2023-09:36:23] [I] Preview Features: Use default preview flags.
[09/19/2023-09:36:23] [I] Input(s): fp16:chw
[09/19/2023-09:36:23] [I] Input(s): fp16:chw
[09/19/2023-09:36:23] [I] Output(s): fp16:chw
[09/19/2023-09:36:23] [I] Output(s): fp16:chw
[09/19/2023-09:36:23] [I] Input build shapes: model
[09/19/2023-09:36:23] [I] Input calibration shapes: model
[09/19/2023-09:36:23] [I] === System Options ===
[09/19/2023-09:36:23] [I] Device: 0
[09/19/2023-09:36:23] [I] DLACore: 
[09/19/2023-09:36:23] [I] Plugins:
[09/19/2023-09:36:23] [I] === Inference Options ===
[09/19/2023-09:36:23] [I] Batch: Explicit
[09/19/2023-09:36:23] [I] Input inference shapes: model
[09/19/2023-09:36:23] [I] Iterations: 10
[09/19/2023-09:36:23] [I] Duration: 3s (+ 200ms warm up)
[09/19/2023-09:36:23] [I] Sleep time: 0ms
[09/19/2023-09:36:23] [I] Idle time: 0ms
[09/19/2023-09:36:23] [I] Streams: 1
[09/19/2023-09:36:23] [I] ExposeDMA: Disabled
[09/19/2023-09:36:23] [I] Data transfers: Enabled
[09/19/2023-09:36:23] [I] Spin-wait: Disabled
[09/19/2023-09:36:23] [I] Multithreading: Disabled
[09/19/2023-09:36:23] [I] CUDA Graph: Disabled
[09/19/2023-09:36:23] [I] Separate profiling: Enabled
[09/19/2023-09:36:23] [I] Time Deserialize: Disabled
[09/19/2023-09:36:23] [I] Time Refit: Disabled
[09/19/2023-09:36:23] [I] NVTX verbosity: 2
[09/19/2023-09:36:23] [I] Persistent Cache Ratio: 0
[09/19/2023-09:36:23] [I] Inputs:
[09/19/2023-09:36:23] [I] === Reporting Options ===
[09/19/2023-09:36:23] [I] Verbose: Enabled
[09/19/2023-09:36:23] [I] Averages: 10 inferences
[09/19/2023-09:36:23] [I] Percentiles: 90,95,99
[09/19/2023-09:36:23] [I] Dump refittable layers:Disabled
[09/19/2023-09:36:23] [I] Dump output: Disabled
[... 563 lines cut ...]
[09/19/2023-09:36:27] [V] [TRT] Searching for input: onnx::QuantizeLinear_496
[09/19/2023-09:36:27] [V] [TRT] QuantizeLinear_3 [QuantizeLinear] inputs: [inputs -> (18, 3, 256, 704)[FLOAT]], [onnx::QuantizeLinear_495 -> ()[FLOAT]], [onnx::QuantizeLinear_496 -> ()[INT8]], 
[09/19/2023-09:36:27] [E] [TRT] ModelImporter.cpp:726: While parsing node number 1 [QuantizeLinear -> "onnx::DequantizeLinear_497"]:
[09/19/2023-09:36:27] [E] [TRT] ModelImporter.cpp:727: --- Begin node ---
[09/19/2023-09:36:27] [E] [TRT] ModelImporter.cpp:728: input: "inputs"
input: "onnx::QuantizeLinear_495"
input: "onnx::QuantizeLinear_496"
output: "onnx::DequantizeLinear_497"
name: "QuantizeLinear_3"
op_type: "QuantizeLinear"

[09/19/2023-09:36:27] [E] [TRT] ModelImporter.cpp:729: --- End node ---
[09/19/2023-09:36:27] [E] [TRT] ModelImporter.cpp:731: ERROR: builtin_op_importers.cpp:1216 In function QuantDequantLinearHelper:
[6] Assertion failed: scaleAllPositive && "Scale coefficients must all be positive"
[09/19/2023-09:36:27] [E] Failed to parse onnx file
[09/19/2023-09:36:27] [I] Finish parsing network model
[09/19/2023-09:36:27] [E] Parsing model failed
[09/19/2023-09:36:27] [E] Failed to create engine from model or file.
[09/19/2023-09:36:27] [E] Engine set up failed
&&&& FAILED TensorRT.trtexec [TensorRT v8502] # trtexec --onnx=model/resnet50int8/camera.backbone.onnx --fp16 --int8 --inputIOFormats=fp16:chw,fp16:chw, --outputIOFormats=fp16:chw,fp16:chw, --saveEngine=model/resnet50int8/build/camera.backbone.plan --memPoolSize=workspace:2048 --verbose --dumpLayerInfo --dumpProfile --separateProfileRun --profilingVerbosity=detailed --exportLayerInfo=model/resnet50int8/build/camera.backbone.json
```

How can I solve it?

Thank you

sangjinpark97, 2023-09-21:
It is caused from the process of customization. Solved.

closed 2023-09-21 by sangjinpark97

zhangshen12356, 2024-04-24:
> It is caused by the process of customization. Solved.

I met the problem too, how did you solve it?