NVIDIA-AI-IOT/Lidar_AI_Solution #185: AttributeError: 'ConfigDict' object has no attribute 'data'

qq160816, 2023-09-20:
I run "python tool/dump-data.py" output error.

Please tell me why?

Traceback (most recent call last):
  File "tool/dump-data.py", line 97, in <module>
    dump_tensor(args)
  File "tool/dump-data.py", line 49, in dump_tensor
    dataset = build_dataset(cfg.data.test)
  File "/home/wangweilin/anaconda3/envs/bevfusion_mit/lib/python3.8/site-packages/mmcv/utils/config.py", line 507, in __getattr__
    return getattr(self._cfg_dict, name)
  File "/home/wangweilin/anaconda3/envs/bevfusion_mit/lib/python3.8/site-packages/mmcv/utils/config.py", line 48, in __getattr__
    raise ex
AttributeError: 'ConfigDict' object has no attribute 'data'

sangjinpark97, 2023-09-21:
Literally config file you use has no attribute 'data'.
You need to check the config argument.

qq160816, 2023-10-08:
> Literally config file you use has no attribute 'data'. You need to check the config argument.

I used c++ to save data is ok.
Thank you!

closed 2023-10-08 by qq160816