NVIDIA-AI-IOT/Lidar_AI_Solution #186: Assert failed 💀. weight_scales_.size() == bias.numel in file src/spconv/engine.cu:993, message: Invalid weight scales 1 != 16

sangjinpark97, 2023-09-25:
Hi, when I run `bash tool/run.sh`, assertion error below raises.

```shell
Assert failed 💀. weight_scales_.size() == bias.numel in file src/spconv/engine.cu:993, message: Invalid weight scales 1 != 16
```

How can I solve it?

Thank you !

hopef, 2023-09-26:
Could you upload the ONNX file here?

sangjinpark97, 2023-09-26:
@hopef

This is my [lidar.backbone.xyz.onnx](https://drive.google.com/file/d/1-Ew1bpA3f9-uR6saAtj6hCg8vOQV4tZ6/view?usp=drive_link)

Thank you

hopef, 2023-09-26:
[image]
The layer output_precision is int8, but the weight_dynamic_ranges is NaN. That is not the expected behavior. This means:
1. If you want to run at INT8, the [PTQ](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/qat/ptq.py) is needed. That will generate the correct weight_dynamic_ranges.
2. If you want to run at FP16, you can comment out the code below:
  qat/export-scn.py: line62-line64
```python
    # for name, module in model.named_modules():
        # module.precision = "int8"
        # module.output_precision = "int8"
```

sangjinpark97, 2023-09-26:
@hopef Thank you very much!

I'm using PTQ model from ResNet50 + VoxelNet. My `environment.sh` is as follows.

```shell
export DEBUG_MODEL=resnet50int8

export DEBUG_PRECISION=fp16
export DEBUG_DATA=custom-data
export USE_Python=ON
```

As you said, if I want to run at FP16, I don't need to generate PTQ model (or DO NOT generate or it doesn't matter?) ?

sangjinpark97, 2023-09-26:
@hopef Sorry to keep asking

When I run `export-scn.py` after commenting the lines you pointed, The error log is shown.

```shell
Tracing model inference...
> Do inference...
   --> SparseConvolutionQunat0[subm] -> Input 0, Output 1
Traceback (most recent call last):
  File "qat/export-scn-ptq.py", line 70, in <module>
    exptool.export_onnx(model, voxels, coors, batch_size, inverse_indices, args.save)
  File "/home/sensor_fusion/Lidar_AI_Solution/CUDA-BEVFusion_mmdet3d/qat/lean/exptool.py", line 413, in export_onnx
    y = model(voxels, coors, batch_size)
  File "/home/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/sensor_fusion/Lidar_AI_Solution/CUDA-BEVFusion_mmdet3d/qat/lean/exptool.py", line 288, in impl
    x = self.conv_input(input_sp_tensor)
  File "/home/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/sensor_fusion/Lidar_AI_Solution/CUDA-BEVFusion_mmdet3d/qat/lean/exptool.py", line 64, in internal_forward
    bind_fn(self, ilayer, y, *args)
  File "/home/sensor_fusion/Lidar_AI_Solution/CUDA-BEVFusion_mmdet3d/qat/lean/exptool.py", line 184, in symbolic_sparse_convolution_quant
    output_precision = "fp16" if hasattr(self, "output_precision") is None else self.output_precision
  File "/home/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1185, in __getattr__
    raise AttributeError("'{}' object has no attribute '{}'".format(
AttributeError: 'SparseConvolutionQunat' object has no attribute 'output_precision'
```

So I changed the commented line to this

```python
    for name, module in model.named_modules():
        module.precision = "fp16"
        module.output_precision = "fp16"
```

but same error related to `weight_scales` occurs.

hopef, 2023-09-26:
Sorry, the set to fp16 is correct.
```python
  for name, module in model.named_modules():
        module.precision = "fp16"
        module.output_precision = "fp16"
```

Could you upload the latest(precision=fp16) ONNX file here?

hopef, 2023-09-26:
Actually, you can export the int8 model, but use fp16 for inference.

sangjinpark97, 2023-09-26:
This is my [latest ONNX](https://drive.google.com/file/d/1ChzEanqHb1NVOpa7zpZXpM4efS2pH03s/view?usp=sharing)

sangjinpark97, 2023-09-26:
@hopef , I found that `input_quantizer.amax' and `weight_quantizer.amax` are NaN.

Can I get some advice about it?

hopef, 2023-09-26:
The weight_dynamic_ranges check only occurs in layer.compute_precision==int8.  and I can load your onnx correctly.
Please double-check it?

sangjinpark97, 2023-09-26:
@hopef, Thanks I successfully load my model !

But, the prediction results is not accurate ;)
I'll check the reason. Anyway, thank you! 

Meanwhile, when I use the INT8 onnx model and run `build_trt_engine.sh`, the error posted in [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/issues/182) occurs. 

When I visualize the camera backbone onnx, all layers' scales are nan..

sangjinpark97, 2023-10-10:
It was due to my customization of model. Closed

closed 2023-10-10 by sangjinpark97