NVIDIA-AI-IOT/Lidar_AI_Solution #187: how to convert nv::Tensor lidar_points to pcl::PointCloud<pcl::PointXYZI>

artofstate, 2023-09-25:
Hi! Have u tried ROS to finish visualization ? I don't know how to convert them correctly.

hopef, 2023-09-26:
Nv::Tensor is just a data structure that stores a matrix. You can get its raw data pointer. There is nothing special about it.

artofstate, 2023-09-26:
> Nv::Tensor is just a data structure that stores a matrix. You can get its raw data pointer. There is nothing special about it.
Sorry, Could you please show me in detail? How to get the position of every point with nv::Tensor type?
In Cuda_BEVFusion/src/main.cpp, auto lidar_points = nv::Tensor::load(nv::format("%s/points.tensor", data), false); Then I'v tried std::cout<<*(unsigned short*)lidar_pointes.ptr<unsigned short>()<<std::endl;  I got a strange output.