NVIDIA-AI-IOT/Lidar_AI_Solution #188: how to apply CUDA-BEVFusion to fit our own model

YueSun0609, 2023-09-26:
Hi,

I used the SwinTransformer as my camera.encoder.backbone, which is different from yours used in CUDA-BEVFusion. May I ask what should I do to correctly export my onnx models? I exported without errors, but when I use d test-mAP-for-cuda.py to print my mAP, the inference time is not shorter and mAP is much lower. Could you help me or give me some clues? Thank you so much!!!

hopef, 2023-09-27:
I also found that. the SwinTransformer has a longer inference time(~68 ms) on ORIN.
Additionally, SwinTransformer is much easier to trigger NaN output on TensorRT. You need to be careful and double-check.

YueSun0609, 2023-10-09:
> I also found that. the SwinTransformer has a longer inference time(~68 ms) on ORIN. Additionally, SwinTransformer is much easier to trigger NaN output on TensorRT. You need to be careful and double-check.

Thank you for your reply. I finally found the error is that I calculated the inference time in a wrong way.

closed 2023-10-09 by YueSun0609