NVIDIA-AI-IOT/Lidar_AI_Solution #190: incorrect inference on KITTI dataset

sangjinpark97, 2023-10-05:
Hi, I trained BEVFusion on KITTI dataset and I got a reasonable performance with the naive torch model.
I have successfully built FP16 and INT8 models.
But both models produce the incorrect inference.
For example,
[image]

Can you check my onnx models?
You can download my models at this [link](https://drive.google.com/file/d/1J31ljJ2yEhLjY2C-7c8t2GqukLPhjL6b/view?usp=sharing).

I would really appreciate any help you can provide.

sangjinpark97, 2023-10-10:
I found that the problem is the input data type by using `xxxx_ptq.pth` generated after `ptq.py` during inference. 
When using `xxx_ptq.pth`, the results get destroyed with fp16 data, on the other hand, the results are good with fp32 data.

But, I cannot use the fp32 input with `libpybev`. How can I love?

crystal250, 2024-01-26:
Hello, how did you modify the configuration file in mit-bevfusion to train the model and convert it to an onnx model?

eugenebak, 2024-05-22:
@crystal250 Yes, I've modified many parts of config in mit-bevfusion according to KITTI dataset. This was not trivial.

closed 2024-05-22 by eugenebak