NVIDIA-AI-IOT/Lidar_AI_Solution #193: My mAP is much lower

YueSun0609, 2023-10-09:
Hi ~
I tried to reproduce the whole process to export a pytorch model (which is bevfusion-det.pth) to onnx/plan models and then use test-mAP-for-cuda.py to calculate the mAP, but found my mAP is much lower. Could you help me with that? Thank you so much!

My steps are:
1. run the following to convert it to get bevfusion_ptq.pth first:
```
python qat/ptq.py \
  --config=bevfusion/configs/nuscenes/det/transfusion/secfpn/camera+lidar/resnet50/convfuser.yaml \
  --ckpt=model/resnet50/bevfusion-det.pth
```
2. export 5 onnx models by:
```
python qat/export-camera.py
python qat/export-transfuser.py 
python qat/export-scn.py 
```
3. change sth. in environment.sh as:
```
DEBUG_MODEL=resnet50-int8
export DEBUG_PRECISION=int8
USE_Python=ON
```
4. run
```
. tool/environment.sh
bash tool/build_trt_engine.sh
bash tool/run.sh
python tool/pybev.py
```
5. then run test-mAP-for-cuda.py and I got:
[image]

YueSun0609, 2023-10-09:
Even I used onnx models downloaded from [https://drive.google.com/file/d/1bPt3D07yyVuSuzRAHySZVR2N15RqGHHN/view?usp=sharing](url) and then run above step 3-5 got the same result.

hopef, 2023-10-20:
You can check the output. If NaN is present it means that you may be encountering bugs in TensorRT for Transformer. You may need to upgrade the TensorRT.

YueSun0609, 2023-10-20:
> You can check the output. If NaN is present it means that you may be encountering bugs in TensorRT for Transformer. You may need to upgrade the TensorRT.

Do you mean nan in "per -class results"? As far as I remember that's because the code sets the value there, eg. , traffic_cone has no orientation.

YueSun0609, 2023-10-20:
> You can check the output. If NaN is present it means that you may be encountering bugs in TensorRT for Transformer. You may need to upgrade the TensorRT.

I checked my tensorrt, the version is:
[image]

hopef, 2023-10-20:
I mean the TensorRT predictions. like bounding boxes, etc.

YueSun0609, 2023-11-01:
> I mean the TensorRT predictions. like bounding boxes, etc.

Hi hopef, I finally achieve the desired mAP after I downgrade my tensorrt from 8.6 to 8.5. Thank you so much!

YueSun0609, 2023-11-01:
Just downgrade my tensorrt from 8.6 to 8.5 solve this problem.

closed 2023-11-01 by YueSun0609