NVIDIA-AI-IOT/Lidar_AI_Solution #197: how to use DLA and GPU to accelerate？

leslie27ch, 2023-10-10:
[image]
1. hello,the picture show inference use GPU+DLA，but i donot find where to use DLA？

2. when let cam_backbone.onnx use DLACORE 0 and fuser.onnx use DLACORE 1， they could be built successful，but when running 的 engine，happened the error：
[image]
[image]

[image]

leslie27ch, 2023-10-10:
thank you  very much for giving some inspirationor information about how to use dla and gpu simultaneously in the project to accelarate .

leslie27ch, 2023-10-10:
thank you very much for giving some inspiration or information about how to use dla and gpu simultaneously in the project to accelarate .

hopef, 2023-10-19:
DLA requires some special operations. For example: [Deep-Learning-Accelerator-SW](https://github.com/NVIDIA/Deep-Learning-Accelerator-SW)