NVIDIA-AI-IOT/Lidar_AI_Solution #204: Question about output_bound

lizhipengcs, 2023-10-20:
Hi! Thanks for your great works.
I have some questions about the attribute `output_bound ` in the node `SparseConvolution` when exporting model to onnx file. What is the meaning of `output_bound` and How to set the value of `output_bound` ? Would the value of `output_bound` affect the output shape of `submanifold convolution`? From my perspective, the output shape of `submanifold convolution` depends on the input shape. So the `output_bound` only takes effect when `subm=False,` and should not affect the output shape of `submanifold convolution`. Am I right?

`    output_bound = 200000
    if hasattr(self, "output_bound"):
        output_bound = self.output_bound

    nodes.append(
        helper.make_node(
            "SparseConvolution", inputs, [get_tensor_id(y)], f"conv{ilayer}", 
            ndim = self.ndim,
            input_spatial_shape = x.spatial_shape,
            output_spatial_shape = y.spatial_shape,
            in_channels = self.in_channels,
            out_channels = self.out_channels,
            kernel_size = self.kernel_size,
            output_bound = output_bound,
            stride = self.stride,
            dilation = self.dilation,
            padding = self.padding,
            transposed = self.transposed,
            inverse = self.inverse,
            output_padding = self.output_padding,
            groups = self.groups,
            subm = self.subm,
            rulebook = self.indice_key,
            activation = getattr(self, "act_type", "None"),
            input_shape  = x.features.shape,
            output_shape = y.features.shape
        )
    )`

hopef, 2023-10-24:
Yes, you are right.

The `output_bound` is used in spatially sparse conv only which will ignore in submanifold conv.

`output_bound` will apply in the spatially sparse conv rulebook with the below code:
`num_output_real_ = min(num_output_real_, output_bound_);`

lizhipengcs, 2023-10-25:
Thank you for your reply!

closed 2023-10-25 by lizhipengcs