NVIDIA-AI-IOT/Lidar_AI_Solution #207: pytorch_quantization/cuda_ext.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN3c1015SmallVectorBaseIjE8grow_podEPvmm

Charmve, 2023-10-26:
the error log:
```
root@d1a0e04b40c1:/Lidar_AI_Solution/CUDA-BEVFusion# python qat/export-camera.py --ckpt=model/resnet50int8/bevfusion_ptq.pth
Traceback (most recent call last):
  File "qat/export-camera.py", line 40, in <module>
    from pytorch_quantization.nn.modules.tensor_quantizer import TensorQuantizer
  File "/opt/conda/lib/python3.8/site-packages/pytorch_quantization/__init__.py", line 20, in <module>
    from .quant_modules import *
  File "/opt/conda/lib/python3.8/site-packages/pytorch_quantization/quant_modules.py", line 23, in <module>
    from pytorch_quantization import nn as quant_nn
  File "/opt/conda/lib/python3.8/site-packages/pytorch_quantization/nn/__init__.py", line 19, in <module>
    from pytorch_quantization.nn.modules.tensor_quantizer import *
  File "/opt/conda/lib/python3.8/site-packages/pytorch_quantization/nn/modules/tensor_quantizer.py", line 24, in <module>
    from pytorch_quantization.tensor_quant import QuantDescriptor, tensor_quant, fake_tensor_quant, scaled_e4m3
  File "/opt/conda/lib/python3.8/site-packages/pytorch_quantization/tensor_quant.py", line 28, in <module>
    from pytorch_quantization import cuda_ext
ImportError: /opt/conda/lib/python3.8/site-packages/pytorch_quantization/cuda_ext.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN3c1015SmallVectorBaseIjE8grow_podEPvmm
```

some try:
```
root@d1a0e04b40c1:/Lidar_AI_Solution/CUDA-BEVFusion# pip uninstall pytorch_quantization
root@d1a0e04b40c1:/Lidar_AI_Solution/CUDA-BEVFusion# pip install pytorch_quantization
Collecting pytorch_quantization
  Using cached pytorch-quantization-2.2.0.tar.gz (6.8 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [16 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-yxsyqxxd/pytorch-quantization_b7d5804e946d41759b83610b4467a0de/setup.py", line 137, in <module>
          raise RuntimeError(open("ERROR.txt", "r").read())
      RuntimeError:
      ###########################################################################################
      The package you are trying to install is only a placeholder project on PyPI.org repository.
      This package is hosted on NVIDIA Python Package Index.
      
      This package can be installed as:
      ```
      $ pip install --no-cache-dir --extra-index-url https://pypi.nvidia.com pytorch-quantization
      ```
      ###########################################################################################
      
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
root@d1a0e04b40c1:/Lidar_AI_Solution/CUDA-BEVFusion# pip install --no-cache-dir --extra-index-url https://pypi.nvidia.com pytorch-quantization
Looking in indexes: https://pypi.org/simple, https://pypi.nvidia.com
Collecting pytorch-quantization
  Downloading https://pypi.nvidia.com/pytorch-quantization/pytorch_quantization-2.2.0-cp38-cp38-linux_x86_64.whl (2.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.9/2.9 MB 774.4 kB/s eta 0:00:00
Requirement already satisfied: numpy in /opt/conda/lib/python3.8/site-packages (from pytorch-quantization) (1.23.0)
Requirement already satisfied: absl-py>=0.7.0 in /opt/conda/lib/python3.8/site-packages (from pytorch-quantization) (2.0.0)
Requirement already satisfied: scipy in /opt/conda/lib/python3.8/site-packages (from pytorch-quantization) (1.10.1)
Requirement already satisfied: sphinx-glpi-theme in /opt/conda/lib/python3.8/site-packages (from pytorch-quantization) (0.3)
Requirement already satisfied: prettytable in /opt/conda/lib/python3.8/site-packages (from pytorch-quantization) (3.9.0)
Requirement already satisfied: pyyaml in /opt/conda/lib/python3.8/site-packages (from pytorch-quantization) (6.0.1)
Requirement already satisfied: wcwidth in /opt/conda/lib/python3.8/site-packages (from prettytable->pytorch-quantization) (0.2.8)
Installing collected packages: pytorch-quantization
Successfully installed pytorch-quantization-2.2.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
root@d1a0e04b40c1:/Lidar_AI_Solution/CUDA-BEVFusion# 
```

the error has been living:

```
root@d1a0e04b40c1:/Lidar_AI_Solution/CUDA-BEVFusion# python -c "import pytorch_quantization as pq; print(pq.__version__)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/conda/lib/python3.8/site-packages/pytorch_quantization/__init__.py", line 20, in <module>
    from .quant_modules import *
  File "/opt/conda/lib/python3.8/site-packages/pytorch_quantization/quant_modules.py", line 23, in <module>
    from pytorch_quantization import nn as quant_nn
  File "/opt/conda/lib/python3.8/site-packages/pytorch_quantization/nn/__init__.py", line 19, in <module>
    from pytorch_quantization.nn.modules.tensor_quantizer import *
  File "/opt/conda/lib/python3.8/site-packages/pytorch_quantization/nn/modules/tensor_quantizer.py", line 24, in <module>
    from pytorch_quantization.tensor_quant import QuantDescriptor, tensor_quant, fake_tensor_quant, scaled_e4m3
  File "/opt/conda/lib/python3.8/site-packages/pytorch_quantization/tensor_quant.py", line 28, in <module>
    from pytorch_quantization import cuda_ext
ImportError: /opt/conda/lib/python3.8/site-packages/pytorch_quantization/cuda_ext.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN3c1015SmallVectorBaseIjE8grow_podEPvmm
```

hopef, 2023-10-26:
Hi, wei,

You can refer the official site of pytorch_quantization_library to check it.

 
the error log:
some try:

adidier17, 2023-11-09:
I wonder if this is a version issue. @hopef can you share what is the version number of `pytorch_quantization` that you used? 

@Charmve you have downloaded version 2.20, which in the [release history](https://pypi.org/project/pytorch-quantization/#history) was released September 28 2023, but the [latest commit in this repo](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/commits/master/CUDA-BEVFusion) is from July 16. However if you pull the available versions using pip directly
```
pip index --extra-index-url https://pypi.nvidia.com versions pytorch-quantization
WARNING: pip index is currently an experimental command. It may be removed/changed in a future release without prior warning.
pytorch-quantization (2.2.1)
Available versions: 2.2.1, 2.2.0, 2.1.3, 2.1.2
```
version 2.1.2, which is not listed on the site page is available. I do not know its date of release, so perhaps it is correct? @hopef can you confirm if this is the version that was used or is there possibly another older version that is no longer available?

W6WM9M, 2023-11-17:
I encountered the same error with ```pytorch-quantization=2.2.1``` but the error is resolved with ```pytorch-quantization=2.1.3```

bigsheep2018, 2024-01-22:
update:
2.1.3 pre-build version not work.
build from source works.