NVIDIA-AI-IOT/Lidar_AI_Solution #208: Number of lidar sweeps used for CUDA-BEVFusion inference

W6WM9M, 2023-10-26:
Hi, may I check if each CUDA-BEVFusion inference was done using only the current timestamp lidar cloud without additional lidar sweeps from the previous timestamps? If I'm correct, the author of BEVFusion uses 9 additional sweeps during training and inference for better detection performance. However, I assume that using additional sweeps here would lead to a slower inference time right?

hopef, 2023-11-01:
the author of BEVFusion uses 9 additional sweeps during training and inference for better detection performance
-> Where did you get this? Could you share the link with me?

W6WM9M, 2023-11-01:
Under the defined test pipeline of BEVFusion in their [config file](https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/det/default.yaml), the lidar sweeps_num is 9. This means they used an additional 9 sweeps for testing.  

````
test_pipeline:
  -
    type: LoadMultiViewImageFromFiles
    to_float32: true
  -
    type: LoadPointsFromFile
    coord_type: LIDAR
    load_dim: ${load_dim}
    use_dim: ${use_dim}
    reduce_beams: ${reduce_beams}
    load_augmented: ${load_augmented}
  -
    type: LoadPointsFromMultiSweeps
    sweeps_num: 9
    load_dim: ${load_dim}
    use_dim: ${use_dim}
    reduce_beams: ${reduce_beams}
    pad_empty_sweeps: true
    remove_close: true
    load_augmented: ${load_augmented}
````

You can also refer to https://github.com/mit-han-lab/bevfusion/issues/463 for further information on the better detection performance due to the higher number of sweeps.

hopef, 2023-11-16:
I assume that using additional sweeps here would lead to a slower inference time right?
-> No, The times measured here are based on 9 lidar points.

W6WM9M, 2023-11-20:
Noted, thank you for the reply.

closed 2023-11-20 by W6WM9M