NVIDIA-AI-IOT/Lidar_AI_Solution #209: cuOSD stb font backend memory too big

zxpshadow, 2023-10-30:
In order to perform asynchronous operations using multiple contexts, it was found that the OSD physical memory was heavily occupied. However, it was found that the backend of the font was heavily occupied, mainly due to measurement_ Text is occupied, so could you please help confirm

1. Does asynchronous operation require multiple contexts

2. Can multiple contexts share font backend

3. Inaccurate measurement of font backend STB font (font set to 10, return height is 12)

zxpshadow, 2023-10-31:
thanks,  Keep On Waiting

hopef, 2023-11-01:
it was found that the OSD physical memory was heavily occupied
-> Could you expand on this?

hopef, 2023-11-01:
Inaccurate measurement of font backend STB font (font set to 10, return height is 12)
-> The measurement function is designed to return the size of the area needed for the rendering. It may include the background area(like a filled background), not just text.

zxpshadow, 2023-11-13:
> Inaccurate measurement of font backend STB font (font set to 10, return height is 12) -> The measurement function is designed to return the size of the area needed for the rendering. It may include the background area(like a filled background), not just text.

pic 1
[image]
pic 2
[image]

Sorry for the delay in other matters last week and not responding in a timely manner。
pic 1 shows the actual drawn font height of approximately 19 pixels
pic 2 shows that the font height returned by cuosd_measure_text is 10 pixels
My understanding is cuosd_ Measure_ Text returns the font height occupied by the drawing