NVIDIA-AI-IOT/Lidar_AI_Solution #211: KeyError: ‘LoadRadarPointsMultiSweeps...’

Yunge6666, 2023-11-04:
When I run .../Lidar_AI_Solution/CUDA-BEVFusion/bevfusion$ python tools/test-mAP-for-cuda.py, I got the following error.
KeyError: 'LoadRadarPointsMultiSweeps is not in the pipeline registry'
KeyError: "NuScenesDataset: 'LoadRadarPointsMultiSweeps is not in the pipeline registry'"

Anyone solve the such problem, plz help me. Thanks~

hopef, 2023-11-04:
You can try the below commands:
```bash
git clone https://github.com/mit-han-lab/bevfusion

cd bevfusion
git checkout db75150717a9462cb60241e36ba28d65f6908607
```

Yunge6666, 2023-11-11:
> You can try the below commands:
> 
> ```shell
> git clone https://github.com/mit-han-lab/bevfusion
> 
> cd bevfusion
> git checkout db75150717a9462cb60241e36ba28d65f6908607
> ```

It looks like you are providing me with another version of bevfusion, with this version I no longer see the above error, but I encounter a new error which is keyerror:[''location'']. I think this may be some bug in the code regarding the nusc dataset, it happens at line 218 of /bevfusion/mmdet3d/datasets/nuscenes_dataset.py:  location=info["location"], when I change it to location = info.get('location', "singapore-queenstown"), it works, but now I have to specify a map, which greatly affects the results of my evaluation. I don't know if you have a better solution.