NVIDIA-AI-IOT/Lidar_AI_Solution #212: CUDA Runtime error cudaStreamSynchronize(stream) # an illegal memory access was encountered, code = cudaErrorIllegalAddress [ 700 ] in file src/spconv/engine.cu:1212

bujianyiwang, 2023-11-06:


hopef, 2023-11-06:
#210

bujianyiwang, 2023-11-10:
Hi,i have trained my model and qat with int8 and buile trt engine, when i deploy model adapting to framework,i am confused to some parameters.
my env:
`3090
CUDA11.4
TensorRT8.5.2`
lidar-voxelization.cu
`virtual unsigned int indices_dim() override { return 4; }`
engine.hpp
`/**
    Inference function for sparse convolution

    features_shape: The shape of the input feature matrix, it must be two elements.
    features_dtype: The data type of the input feature matrix, it must be Float16 now.
    features_data:  The data pointer of the input feature matrix
    indices_shape:  The shape of the input indices matrix, it must be two elements[n, 4]
    indices_dtype:  The data type of the input indices matrix, it must be Int32 now.
    indices_data:   The data pointer of the input indices matrix
    batch:          The batch size of the input, it must be 1 now.
    grid_size:      The grid size of the input data, For example: 41,1440,1440 or 1440,1440,41
    stream:         Which stream is expected to enqueue the inference.
  **/
  Exported virtual DTensor* forward(const std::vector<int64_t>& features_shape,
                                    DType features_dtype, void* features_data,
                                    const std::vector<int64_t>& indices_shape, DType indices_dtype,
                                    void* indices_data, int batch, std::vector<int> grid_size,
                                    void* stream = nullptr) = 0;`
I cannot understand indices_dtype, it seems like enumerate type but i cannot get source code to read the definition.
Can you explain that to me?
thx
And my repo is failed with the output below:
`CUDA Runtime error cudaStreamSynchronize(stream) # an illegal memory access was encountered, code = cudaErrorIllegalAddress [ 700 ] in file src/spconv/engine.cu:1212`

closed 2023-11-10 by bujianyiwang