NVIDIA-AI-IOT/Lidar_AI_Solution #215: Deployment of lidar-only model

imurfutur3, 2023-11-13:
Thank you very much for your outstanding work. However, I encountered some problems while deploying the lidar-only model. I am using the lidar/voxelnet_0p075.yaml config. Accordingly, I have only exported the 3D backbone and head of the lidar as two separate ONNX files. Regarding the C++ part, I skipped the fuser and directly passed the output of the 3D backbone to the head. The program runs, but the results are incorrect. Do you know what the issue might be?

`const nvtype::half* lidar_feature = this->lidar_scn_->forward(lidar_points_device_, num_points, stream);
    // nvtype::half* normed_images = (nvtype::half*)camera_images;
    // if (do_normalization) {
    //   normed_images = (nvtype::half*)this->normalizer_->forward((const unsigned char**)(camera_images), stream);
    // }
    // const nvtype::half* depth = this->camera_depth_->forward(lidar_points_device_, num_points, 5, stream);

    // this->camera_backbone_->forward(normed_images, depth, stream);
    // const nvtype::half* camera_bev = this->camera_bevpool_->forward(
    //     this->camera_backbone_->feature(), this->camera_backbone_->depth(), this->camera_geometry_->indices(),
    //     this->camera_geometry_->intervals(), this->camera_geometry_->num_intervals(), stream);

    // const nvtype::half* camera_bevfeat = camera_vtransform_->forward(camera_bev, stream);
    // const nvtype::half* fusion_feature = this->transfusion_->forward(camera_bevfeat, lidar_feature, stream);
    const nvtype::half* fusion_feature = this->transfusion_->forward(lidar_feature, stream);
    return this->transbbox_->forward(fusion_feature, param_.transbbox.confidence_threshold, stream,
                                     param_.transbbox.sorted_bboxes);
`

BR,
Joyn

imurfutur3, 2023-11-13:
Solved.

closed 2023-11-13 by imurfutur3

MarioUncle-Wang, 2024-07-04:
> Solved.

hi, I also encountered the same problem, 
How did you solve this problem?