NVIDIA-AI-IOT/Lidar_AI_Solution #216: function and parameters in lidar-voxelization.cu function and engine.hpp

bujianyiwang, 2023-11-15:
/**
    Inference function for sparse convolution

    features_shape: The shape of the input feature matrix, it must be two elements.
    features_dtype: The data type of the input feature matrix, it must be Float16 now.
    features_data:  The data pointer of the input feature matrix
    indices_shape:  The shape of the input indices matrix, it must be two elements[n, 4]
    indices_dtype:  The data type of the input indices matrix, it must be Int32 now.
    indices_data:   The data pointer of the input indices matrix
    batch:          The batch size of the input, it must be 1 now.
    grid_size:      The grid size of the input data, For example: 41,1440,1440 or 1440,1440,41
    stream:         Which stream is expected to enqueue the inference.
  **/
`Exported virtual DTensor* forward(const std::vector<int64_t>& features_shape,
                                    DType features_dtype, void* features_data,
                                    const std::vector<int64_t>& indices_shape, DType indices_dtype,
                                    void* indices_data, int batch, std::vector<int> grid_size,
                                    void* stream = nullptr) = 0;`
`voxelization_->forward(points, num_points, stream, param_.order);
    native_scn_output_ = native_scn_->forward(
        std::vector<int64_t>{voxelization_->num_voxels(), voxelization_->voxel_dim()}, spconv::DType::Float16,
        (void*)voxelization_->features(), std::vector<int64_t>{voxelization_->num_voxels(), voxelization_->indices_dim()},
        spconv::DType::Int32, (void*)voxelization_->indices(), 1, voxelization_->grid_size(), stream);`

Hello，what does the points dimension order? does it must be x,y,z,intensity,timestamp_us？
I had trained my model, my lidar points has eight dimensions as the order of x,y,z,vx,vy,vz,intensity,timestamp_us, and my test lidar points" order is same! 
But i face that problem below:
[image]

hopef, 2023-11-16:
Could you provide the onnx file and the sample data here?

hopef, 2023-11-16:
This error often occurs when **grid_size** does not match **indices**.
Such as grid_size is 41,1440,1440, but indices give 1400, 1400, 30.

bujianyiwang, 2023-11-16:
> This error often occurs in **ScatterDense** node when **grid_size** does not match **indices**. Such as grid_size is 41,1440,1440, but gives indices 1400, 1400, 30. ScatterDense.format tells us what format should be used.
What grid_size does spconv or ScatterDense support，how about (1760, 1760, 41)？

hopef, 2023-11-16:
The only requirement here is to match, not its size.

closed 2023-11-17 by bujianyiwang