NVIDIA-AI-IOT/Lidar_AI_Solution #218: There is no resnet50 in bevfusion train code

FantineFu, 2023-11-24:
I can't found any resnet configs and support in bevfusion repo, even  in the specific commit db75150717a9462cb60241e36ba28d65f6908607 you provide.
There is no "bevfusion/configs/nuscenes/det/transfusion/secfpn/camera+lidar/resnet50/convfuser.yaml" as you mentioned in ptq, how can I get resnet50 model?

hopef, 2023-11-30:
here here,
https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/configs/nuscenes/det/transfusion/secfpn/camera%2Blidar/resnet50/convfuser.yaml

hopef, 2023-11-30:
`simplify` is not necessary. You can try to comment out the `simplify` function.

FantineFu, 2023-11-30:
I also have tried to not simplify, but the .plan I got from this finally caused to error when run.sh:

`
[libprotobuf FATAL /usr/include/google/protobuf/repeated_field.h:1506] CHECK failed: (index) < (current_size_): 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: (index) < (current_size_): 
tool/run.sh: line 41:  1516 Aborted                 (core dumped)
`
How to align my .pth with your example .pth?

By the way, the .pth save by the train code is different from yours in format. I found that the .pth from bevfusion is save by mmcv function, thus it can't load by `torch.load()` as in your export_onnx code. So I load the trained .pth by the way as in bevfusion/tools/test.py . But why your .pth can be loaded directly? How do you save your .pth in train code? I think this may cause the error I met above.

FantineFu, 2023-12-01:
Thanks much, I've solved.

closed 2023-12-01 by FantineFu

Kai2-n, 2024-09-04:
@FantineFu May I ask what modifications are needed to train a network with resnet50 as the backbone, and what training commands should be inputted on the terminal

Anmidy, 2024-12-16:
> > here here, https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/configs/nuscenes/det/transfusion/secfpn/camera%2Blidar/resnet50/convfuser.yaml
> 
> I trained by this yaml, but when I export camera backbone onnx, the pth I trained by this config caused error in simplify:
> 
> `Traceback (most recent call last): File "qat/export-camera.py", line 177, in <module> main() File "qat/export-camera.py", line 160, in main onnx_simp, check = simplify(onnx_orig) File "/opt/conda/lib/python3.8/site-packages/onnxsim/onnx_simplifier.py", line 184, in simplify model_opt_bytes = C.simplify( RuntimeError: /project/third_party/onnx-optimizer/onnxoptimizer/passes/fuse_bn_into_conv.h:95: modify_conv: Assertion s_iter->elem_type() == bbn_iter->elem_type() && s_iter->elem_type() == m_iter->elem_type() && s_iter->elem_type() == var_iter->elem_type() && s_iter->elem_type() == W_iter->elem_type() failed. `
> 
> the exports for other parts of the .pth all went very well, thus I think there may be something wrong just from the backbone training, but I have no idea.

Hello,did you trained the resnet50 backboen bevfusion-det.pth by this commit db75150717a9462cb60241e36ba28d65f6908607?  Can you successfully transform the bevfusion-det.pth to bevfusion-ptq.pth by qat/ptq.py ? I got a error by running ptq.py ,KeyError: "NuScenesDataset: 'LoadRadarPointsMultiSweeps is not in the pipeline registry'"