NVIDIA-AI-IOT/Lidar_AI_Solution #219: Doubts about “MaxDistance”

Like2021, 2023-11-30:
I would like to know what the `MaxDistance` you used in the visualization means, and what determines it?
```c++
std::vector<nvtype::Float4> lidar2image = {{param_.norm_size / MaxDistance, 0, 0, param_.cx},
                 {0, -param_.norm_size / MaxDistance, 0, param_.cy},
                 {0, 0, 0, 1},
                 {0, 0, 0, 1}};

hopef, 2023-11-30:
```c++
lidar_size          = 1024;             // unit pixel
content_width = lidar_size + 900;  // unit pixel
content_height = 1080;          // unit pixel
norm_size = lidar_size * 0.5f;  // unit pixel
cx = content_width * 0.5f;      // unit pixel
cy = content_height * 0.5f;     // unit pixel
MaxDistance = 50;               // unit meter
lidar2image = {{norm_size / MaxDistance, 0, 0, cx}, 
                  {0, -norm_size / MaxDistance, 0, cy}, 
                  {0, 0, 0, 1}, 
                  {0, 0, 0, 1}}; 
```
When we apply it on points of the point-cloud for projection into the canvas space:
```python
# points is N x 4(x, y, z, 1)
#  N x 4    =  N x 4  @    4 x 4
canvas_points = points @ lidar2image.T

# canvas_point.x = point.x * norm_size / MaxDistance + cx;
# canvas_point.y = -point.y * norm_size / MaxDistance + cy;
# When point.x = 0, point.y = 0;
# We get: canvas_point.x = lidar_size * 0.5f,   canvas_point.y = cy;    # This point(ego) will be in the center of the canvas.
#
# When point.x = 50m, point.y = 50m;
# We get: canvas_point.x = 50m / MaxDistance * lidar_size * 0.5f + cx
#              canvas_point.y = cy - 50m / MaxDistance * lidar_size * 0.5f;
#              canvas_point.x = cx + 512;   # where canvas_width = content_width
#              canvas_point.x = cy - 512;   # where canvas_height = content_height
#              # This point(furthest) will be on the edge of the canvas.
```

So, MaxDistance is used to de-unitize the distance and convert it to a scaling ratio.

Like2021, 2023-11-30:
Thanks for your explanation, I think I understand your explanation. However, I also want to know why the MaxDistance here is 50m, and what variables determine this value?

hopef, 2023-12-01:
It's just a personalized variable. You can set it to your favorite range.

Like2021, 2023-12-13:
Thanks for your answer, this is a great job.

closed 2023-12-13 by Like2021