NVIDIA-AI-IOT/Lidar_AI_Solution #222: About camera-bevpool.cu and camera-geometry.cu

CV-player, 2023-12-07:
when I use my own transformation matrix with four cameras, there is a problem: the value of interval.z was negative, causing output_offset to be negative.
`
#pragma unroll
  for (int j = 0; j < tile_size; j++) {
    unsigned int output_offset = interval.z + (feature_block + j) * out_h * out_w;
    output_bevfeat[output_offset] = accumulate[j];
}
`
May be there is some bugs in **camera-geometry.cu**

hopef, 2023-12-07:
Could you add my WeChat(woshixiwanga) to co-debug this?
I know that some implementations in `camera-geometry.cu` may be hard to understand and need to be careful.
I'll update this part later.

songyonger, 2025-08-29:
I have the same problem with you, how do you solve it? @cn-hezhu @hopef