NVIDIA-AI-IOT/Lidar_AI_Solution #224: Could you specify which exact model of the NVIDIA Jetson Orin series was used in this project?

YueqiLuoCata, 2023-12-11:
Could you specify which exact model of the NVIDIA Jetson Orin series was used to achieve the performance metrics detailed in the Readme? :
    Jetson AGX Orin 64GB
    Jetson AGX Orin Industrial
    Jetson AGX Orin 32GB
    Jetson Orin NX 16GB
    Jetson Orin NX 8GB
    Jetson Orin Nano 8GB
    Jetson Orin Nano 4GB

hopef, 2023-12-12:
Jetson AGX Orin 64GB

hopef, 2024-01-11:
1. if 8GB memory is qualified it， if not how many it need？
-> That would be enough to infer. But it would help if you considered removing some of the system footprint. If you turn CUDNN/cuBLAS off on TensorRT further, you can save even more memory.

2. Jetson Orin Nano 8GB sm=86，Whether it is qualified for the project？ whether must be the AGX orin？thank！
-> Jetson Orin Nano 8GB sm=86 is ok for infer. AGX orin isn't necessary.