NVIDIA-AI-IOT/Lidar_AI_Solution #225: cudaErrorInvalidConfiguration [ 9 ]

wep21, 2023-12-16:
I tried to use spconv library from this repositry, but I encountered cudaErrorInvalidConfiguration as below.
```
[Verb🚩] EngineBuilder: libspconv version is: 1.1.2
[Verb🚩] Use inference precision: Float16
[Verb🚩] Parse input 0
[Verb🚩] add0 use output name 5
[Verb🚩] Ignore relu0 because it fused with add
[Verb🚩] add1 use output name 9
[Verb🚩] Ignore relu1 because it fused with add
[Verb🚩] add2 use output name 14
[Verb🚩] Ignore relu2 because it fused with add
[Verb🚩] add3 use output name 18
[Verb🚩] Ignore relu3 because it fused with add
[Verb🚩] add4 use output name 23
[Verb🚩] Ignore relu4 because it fused with add
[Verb🚩] add5 use output name 27
[Verb🚩] Ignore relu5 because it fused with add
[Verb🚩] add6 use output name 32
[Verb🚩] Ignore relu6 because it fused with add
[Verb🚩] add7 use output name 36
[Verb🚩] Ignore relu7 because it fused with add
[Verb🚩] Node size: 40
[Verb🚩] Parse node conv0 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv0[submanifold][out:Int8], precision: Float16. kernel: 16x3x3x3x5, activation: ReLU, rulebook: subm1
[Verb🚩] Parse node conv1 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv1[submanifold][out:Int8], precision: Int8. kernel: 16x3x3x3x16, activation: ReLU, rulebook: subm1
[Verb🚩] Parse node conv2 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv2[submanifold][out:Int8], precision: Int8. kernel: 16x3x3x3x16, activation: None, rulebook: subm1
[Verb🚩] Parse node add0 [Add], numInput=2, numOutput=1
[Verb🚩] Parse node relu0 [ReLU], numInput=1, numOutput=1
[Verb🚩] A policy that can be fused is used: fuse add0 & relu0
[Verb🚩] Parse node conv3 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv3[submanifold][out:Int8], precision: Int8. kernel: 16x3x3x3x16, activation: ReLU, rulebook: subm1
[Verb🚩] Parse node conv4 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv4[submanifold][out:Int8], precision: Int8. kernel: 16x3x3x3x16, activation: None, rulebook: subm1
[Verb🚩] Parse node add1 [Add], numInput=2, numOutput=1
[Verb🚩] Parse node relu1 [ReLU], numInput=1, numOutput=1
[Verb🚩] A policy that can be fused is used: fuse add1 & relu1
[Verb🚩] Parse node conv5 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv5[spatially][out:Int8], precision: Int8. kernel: 32x3x3x3x16, activation: ReLU, rulebook: spconv1
[Verb🚩] Parse node conv6 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv6[submanifold][out:Int8], precision: Int8. kernel: 32x3x3x3x32, activation: ReLU, rulebook: subm2
[Verb🚩] Parse node conv7 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv7[submanifold][out:Int8], precision: Int8. kernel: 32x3x3x3x32, activation: None, rulebook: subm2
[Verb🚩] Parse node add2 [Add], numInput=2, numOutput=1
[Verb🚩] Parse node relu2 [ReLU], numInput=1, numOutput=1
[Verb🚩] A policy that can be fused is used: fuse add2 & relu2
[Verb🚩] Parse node conv8 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv8[submanifold][out:Int8], precision: Int8. kernel: 32x3x3x3x32, activation: ReLU, rulebook: subm2
[Verb🚩] Parse node conv9 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv9[submanifold][out:Int8], precision: Int8. kernel: 32x3x3x3x32, activation: None, rulebook: subm2
[Verb🚩] Parse node add3 [Add], numInput=2, numOutput=1
[Verb🚩] Parse node relu3 [ReLU], numInput=1, numOutput=1
[Verb🚩] A policy that can be fused is used: fuse add3 & relu3
[Verb🚩] Parse node conv10 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv10[spatially][out:Int8], precision: Int8. kernel: 64x3x3x3x32, activation: ReLU, rulebook: spconv2
[Verb🚩] Parse node conv11 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv11[submanifold][out:Int8], precision: Int8. kernel: 64x3x3x3x64, activation: ReLU, rulebook: subm3
[Verb🚩] Parse node conv12 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv12[submanifold][out:Int8], precision: Int8. kernel: 64x3x3x3x64, activation: None, rulebook: subm3
[Verb🚩] Parse node add4 [Add], numInput=2, numOutput=1
[Verb🚩] Parse node relu4 [ReLU], numInput=1, numOutput=1
[Verb🚩] A policy that can be fused is used: fuse add4 & relu4
[Verb🚩] Parse node conv13 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv13[submanifold][out:Int8], precision: Int8. kernel: 64x3x3x3x64, activation: ReLU, rulebook: subm3
[Verb🚩] Parse node conv14 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv14[submanifold][out:Int8], precision: Int8. kernel: 64x3x3x3x64, activation: None, rulebook: subm3
[Verb🚩] Parse node add5 [Add], numInput=2, numOutput=1
[Verb🚩] Parse node relu5 [ReLU], numInput=1, numOutput=1
[Verb🚩] A policy that can be fused is used: fuse add5 & relu5
[... 75 lines cut ...]
[Verb🚩] Ignore the layer[conv18] output precision[int8] because the model precision configured fp16, so that will reset to fp16
[Verb🚩] Ignore the layer[conv19] precision[int8] because the model precision configured fp16, so that will reset to fp16
[Verb🚩] Ignore the layer[conv19] output precision[int8] because the model precision configured fp16, so that will reset to fp16
[Verb🚩] Ignore the layer[add7] precision[int8] because the model precision configured fp16, so that will reset to fp16
[Verb🚩] Ignore the layer[add7] output precision[int8] because the model precision configured fp16, so that will reset to fp16
[Verb🚩] Ignore the layer[conv20] precision[int8] because the model precision configured fp16, so that will reset to fp16
[Verb🚩] [scatter0]+[Transpose] output shape is 1 x 128 x 2 x 180 x 180
[Verb🚩] [Load model is done.. 🤗]
[Verb🚀] Forward[conv0:Float16] SparseConvolution, input0=17748x5[41x1440x1440]Float16 | output0=17748x16[41x1440x1440]Float16
[Verb🚀] Forward[conv1:Float16] SparseConvolution, input0=17748x16[41x1440x1440]Float16 | output0=17748x16[41x1440x1440]Float16
[Verb🚀] Forward[conv2:Float16] SparseConvolution, input0=17748x16[41x1440x1440]Float16 | output0=17748x16[41x1440x1440]Float16
[Verb🚀] [add0] Execute add & relu fusion kernel, numel is 283968.
[Verb🚀] Forward[add0:Float16] Add, input0=17748x16[41x1440x1440]Float16,input1=17748x16[41x1440x1440]Float16 | output0=17748x16[41x1440x1440]Float16
[Verb🚀] Forward[conv3:Float16] SparseConvolution, input0=17748x16[41x1440x1440]Float16 | output0=17748x16[41x1440x1440]Float16
[Verb🚀] Forward[conv4:Float16] SparseConvolution, input0=17748x16[41x1440x1440]Float16 | output0=17748x16[41x1440x1440]Float16
[Verb🚀] [add1] Execute add & relu fusion kernel, numel is 283968.
[Verb🚀] Forward[add1:Float16] Add, input0=17748x16[41x1440x1440]Float16,input1=17748x16[41x1440x1440]Float16 | output0=17748x16[41x1440x1440]Float16
CUDA Runtime error gemmfn<<<grid, block, memsize * num_warp, stream>>>( book->route(), book->route_mask(), book->relocation(), (half *)bias_, fp16_input_ptr, fp16output_.ptr(), (half *)weight_, prob.Ci, M, N, K, prob.kv, output_scale_, false) # invalid configuration argument, code = cudaErrorInvalidConfiguration [ 9 ] in file src/spconv/implicit-gemm.cu:165
fish: Job 1, './install/bin/spconv_trt --pts_…' terminated by signal SIGABRT (Abort)
```
The error can be reproduced in [my demo repository](https://github.com/wep21/spconv_trt).
My environment is 
OS: Ubuntu 22.04
GPU: NVIDIA GeForce RTX 3060
Is this a problem with sm architecture?

hopef, 2023-12-20:
[image]

wep21, 2023-12-20:
```
[Verb🚩] EngineBuilder: libspconv version is: 1.1.2
[Verb🚩] Use inference precision: Float16
[Verb🚩] Parse input 0
[Verb🚩] add0 use output name 5
[Verb🚩] Ignore relu0 because it fused with add
[Verb🚩] add1 use output name 9
[Verb🚩] Ignore relu1 because it fused with add
[Verb🚩] add2 use output name 14
[Verb🚩] Ignore relu2 because it fused with add
[Verb🚩] add3 use output name 18
[Verb🚩] Ignore relu3 because it fused with add
[Verb🚩] add4 use output name 23
[Verb🚩] Ignore relu4 because it fused with add
[Verb🚩] add5 use output name 27
[Verb🚩] Ignore relu5 because it fused with add
[Verb🚩] add6 use output name 32
[Verb🚩] Ignore relu6 because it fused with add
[Verb🚩] add7 use output name 36
[Verb🚩] Ignore relu7 because it fused with add
[Verb🚩] Node size: 40
[Verb🚩] Parse node conv0 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv0[submanifold][out:Int8], precision: Float16. kernel: 16x3x3x3x5, activation: ReLU, rulebook: subm1
[Verb🚩] Parse node conv1 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv1[submanifold][out:Int8], precision: Int8. kernel: 16x3x3x3x16, activation: ReLU, rulebook: subm1
[Verb🚩] Parse node conv2 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv2[submanifold][out:Int8], precision: Int8. kernel: 16x3x3x3x16, activation: None, rulebook: subm1
[Verb🚩] Parse node add0 [Add], numInput=2, numOutput=1
[Verb🚩] Parse node relu0 [ReLU], numInput=1, numOutput=1
[Verb🚩] A policy that can be fused is used: fuse add0 & relu0
[Verb🚩] Parse node conv3 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv3[submanifold][out:Int8], precision: Int8. kernel: 16x3x3x3x16, activation: ReLU, rulebook: subm1
[Verb🚩] Parse node conv4 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv4[submanifold][out:Int8], precision: Int8. kernel: 16x3x3x3x16, activation: None, rulebook: subm1
[Verb🚩] Parse node add1 [Add], numInput=2, numOutput=1
[Verb🚩] Parse node relu1 [ReLU], numInput=1, numOutput=1
[Verb🚩] A policy that can be fused is used: fuse add1 & relu1
[Verb🚩] Parse node conv5 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv5[spatially][out:Int8], precision: Int8. kernel: 32x3x3x3x16, activation: ReLU, rulebook: spconv1
[Verb🚩] Parse node conv6 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv6[submanifold][out:Int8], precision: Int8. kernel: 32x3x3x3x32, activation: ReLU, rulebook: subm2
[Verb🚩] Parse node conv7 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv7[submanifold][out:Int8], precision: Int8. kernel: 32x3x3x3x32, activation: None, rulebook: subm2
[Verb🚩] Parse node add2 [Add], numInput=2, numOutput=1
[Verb🚩] Parse node relu2 [ReLU], numInput=1, numOutput=1
[Verb🚩] A policy that can be fused is used: fuse add2 & relu2
[Verb🚩] Parse node conv8 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv8[submanifold][out:Int8], precision: Int8. kernel: 32x3x3x3x32, activation: ReLU, rulebook: subm2
[Verb🚩] Parse node conv9 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv9[submanifold][out:Int8], precision: Int8. kernel: 32x3x3x3x32, activation: None, rulebook: subm2
[Verb🚩] Parse node add3 [Add], numInput=2, numOutput=1
[Verb🚩] Parse node relu3 [ReLU], numInput=1, numOutput=1
[Verb🚩] A policy that can be fused is used: fuse add3 & relu3
[Verb🚩] Parse node conv10 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv10[spatially][out:Int8], precision: Int8. kernel: 64x3x3x3x32, activation: ReLU, rulebook: spconv2
[Verb🚩] Parse node conv11 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv11[submanifold][out:Int8], precision: Int8. kernel: 64x3x3x3x64, activation: ReLU, rulebook: subm3
[Verb🚩] Parse node conv12 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv12[submanifold][out:Int8], precision: Int8. kernel: 64x3x3x3x64, activation: None, rulebook: subm3
[Verb🚩] Parse node add4 [Add], numInput=2, numOutput=1
[Verb🚩] Parse node relu4 [ReLU], numInput=1, numOutput=1
[Verb🚩] A policy that can be fused is used: fuse add4 & relu4
[Verb🚩] Parse node conv13 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv13[submanifold][out:Int8], precision: Int8. kernel: 64x3x3x3x64, activation: ReLU, rulebook: subm3
[Verb🚩] Parse node conv14 [SparseConvolution], numInput=1, numOutput=1
[Verb🚩] Add SparseConvolution conv14[submanifold][out:Int8], precision: Int8. kernel: 64x3x3x3x64, activation: None, rulebook: subm3
[Verb🚩] Parse node add5 [Add], numInput=2, numOutput=1
[Verb🚩] Parse node relu5 [ReLU], numInput=1, numOutput=1
[Verb🚩] A policy that can be fused is used: fuse add5 & relu5
[Verb🚩] Parse node conv15 [SparseConvolution], numInput=1, numOutput=1
[... 100 lines cut ...]
[Verb🚀] Forward[conv10:Float16] SparseConvolution, input0=27774x32[720x720x21]Float16 | output0=18711x64[360x360x11]Float16
[Verb🚀] Forward[conv11:Float16] SparseConvolution, input0=18711x64[360x360x11]Float16 | output0=18711x64[360x360x11]Float16
[Verb🚀] Forward[conv12:Float16] SparseConvolution, input0=18711x64[360x360x11]Float16 | output0=18711x64[360x360x11]Float16
[Verb🚀] [add4] Execute add & relu fusion kernel, numel is 1197504.
[... cut ...]
[Verb🚀] Forward[conv20:Float16] SparseConvolution, input0=10476x128[180x180x5]Float16 | output0=8815x128[180x180x2]Float16
[Verb🚀] [scatter0] ToDense Tactic is ToDenseAndTranspose01423_XYZ
[Verb🚀] Forward[scatter0:Float16] ToDense, input0=8815x128[180x180x2]Float16 | output0=1x128x2x180x180[1x128x2x180x180]Float16
[Verb🚀] Forward[reshape0:Float16] Reshape, input0=1x128x2x180x180[1x128x2x180x180]Float16 | output0=1x256x180x180[1x256x180x180]Float16
🙌 Output.shape: 1 x 256 x 180 x 180
```
It works fine. Thanks.

closed 2023-12-20 by wep21