NVIDIA-AI-IOT/Lidar_AI_Solution #229: Running time

super-liuyang, 2024-01-09:
I am using the Jetson AGX Orin 64G with a power consumption of maxn. However, when I tested the examples you provided, it took 50ms to process one image. Are the six images processed in parallel? If not, how do you achieve an output of up to 28fps?

super-liuyang, 2024-01-11:
@hopef   So  my question is whether you process the six images using a parallel method or a serial method. If you use a parallel method, then the processing time is 40ms, right?
[image]

hopef, 2024-01-11:
Yes, we only use a parallel method to process 6 images. 

Because they can be packed into a batch and fed into TRTEngine.

super-liuyang, 2024-01-11:
@hopef   However, my average processing time is 50ms, and my power consumption has reached the maximum. How can I increase the frame rate?

super-liuyang, 2024-01-11:
@hopef  I also want to know if there is a difference in processing speed between the Python interface and the C++ interface.

hopef, 2024-01-11:
Could you provide your running environment details? 
like TensorRT Version, JetsonPackage Version, CUDA Version, and Reported inference latency.

super-liuyang, 2024-01-11:
@hopef  TensorRT Version:8.5.2.2, JetsonPackage Version:5.1, CUDA Version:11.4 ,  Cudnn：8.6.0.166

hopef, 2024-01-11:
`TensorRT-8.6, cuda-11.4 and cudnn8.6` may be the best choice for you.