NVIDIA-AI-IOT/Lidar_AI_Solution #231: Modifying the parameters min_range, max_range, and voxel_size for voxelization results in a decrease in the accuracy of inference.

nainaigetuide, 2024-01-16:
Hello, I am using CUDA-Bevfusion for inference on my dataset, and I want to modify the parameters related to voxelization to adjust the range of the point cloud. I have made the following modifications to the parameters:
  ```
voxelization.min_range = nvtype::Float3(-72.0f, -72.0f, -5.0);
  voxelization.max_range = nvtype::Float3(+72.0f, +72.0f, +3.0);
  voxelization.voxel_size = nvtype::Float3(0.1f, 0.1f, 0.2f); 
```
After adjusting the maximum and minimum range from 54 to 72, I also changed the grid size from 0.075 to 0.1. This is done to ensure that the grid size is (1440, 1440, 41).
Please ignore the top image in the six pictures.
The inference results before modifying the parameters are
[image]

The results after modifying the maximum range to 72 are
[image]

eugenebak, 2024-05-13:
I meet same problem. Have you solved it?

rayxie11, 2024-06-11:
I think there should be more to be considered. I am trying to increase the range as well without extra training. First of all, the camera projection range need to be changed as well to have bigger range. You should probably change to the following:
```
geometry.xbound = nvtype::Float3(-72.0f, 72.0f, 0.4f); 
geometry.ybound = nvtype::Float3(-72.0f, 72.0f, 0.4f);
```
In addition, the transbox parameters need to be changed. I am not confident with the numbers that I have calculated but I will test them tomorrow. Will update this thread if my test is successful.

eugenebak, 2024-06-18:
@rayxie11, any update about it? thank you