NVIDIA-AI-IOT/Lidar_AI_Solution #234: cuda12 support for spconv v1.1.2

wep21, 2024-01-30:
I would like nvidia team to provide also spconv v1.1.2 shared library for cuda12.x.

wep21, 2024-01-30:
cc @qgai

wep21, 2024-01-30:
cc @hopef

hopef, 2024-01-31:
I'll finish it as soon as possible, thanks.

wep21, 2024-03-12:
@hopef Any updates?

scepter914, 2024-03-15:
I also hope this update.

xuyuan, 2024-03-19:
the cuda 12 version has been upload here: https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/libraries/3DSparseConvolution/libspconv/lib/x86_64

however the results are off:

[image]

Comparing to cuda 11 version:

[image]

@hopef could you fix this? thx!

Xinchengzelin, 2024-06-09:
> the cuda 12 version has been upload here: https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/libraries/3DSparseConvolution/libspconv/lib/x86_64
> 
> however the results are off:
> 
> ![cuda-bevfusion_cuda12](https://private-user-images.githubusercontent.com/239020/314138576-9f93aeb3-08b3-4cb7-8280-0cb4d0292b4b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc5MDY0OTIsIm5iZiI6MTcxNzkwNjE5MiwicGF0aCI6Ii8yMzkwMjAvMzE0MTM4NTc2LTlmOTNhZWIzLTA4YjMtNGNiNy04MjgwLTBjYjRkMDI5MmI0Yi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwOVQwNDA5NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMjQ4MjFhNzk3ODNkOGVhNDgyYjA1NTY1MjZlYzI1ZmM0NjhkOWFkZjRmNDdiNGY4ZTU0MDM0MTcyYmY3OTc4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.MkeyImVsO2Bo7_yMS77K9WRkcgvzD0_XBhqzRcCKfqI)
> 
> Comparing to cuda 11 version:
> 
> ![cuda11-bevfusion_r50_fp16](https://private-user-images.githubusercontent.com/239020/314138831-11092105-4e6b-45b5-b004-1d054a1c663c.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc5MDY0OTIsIm5iZiI6MTcxNzkwNjE5MiwicGF0aCI6Ii8yMzkwMjAvMzE0MTM4ODMxLTExMDkyMTA1LTRlNmItNDViNS1iMDA0LTFkMDU0YTFjNjYzYy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwOVQwNDA5NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jOTU0MjA1NzhjZWYyNDMxYTNhZDkxYzEyYTdlMTJjMjc5MWMwNzgxYzcyMDNhMGZmODY1YmNmZTdmZDgzNDdhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.q5OgjLVr9D5ywdOMbhYSNaY_jO0TwZBrN_msWfbu6BE)
> 
> @hopef could you fix this? thx!

Hi, @xuyuan , spconv v1.1.2 only for Cuda12.2?  I tried Cuda12.0,  which show the error:
```
/usr/bin/ld: warning: libprotobuf.so.23, needed by /home/zetlin/Code/github/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /home/zetlin/Code/github/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so: undefined reference to `google::protobuf::RepeatedField<double>::elements() const'
/usr/bin/ld: /home/zetlin/Code/github/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so: undefined reference to `google::protobuf::internal::ComputeUnknownFieldsSize(google::protobuf::internal::InternalMetadata const&, unsigned long, google::protobuf::internal::CachedSize*)'
/usr/bin/ld: /home/zetlin/Code/github/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so: undefined reference to `google::protobuf::internal::AssignDescriptors(google::protobuf::internal::DescriptorTable const*, bool)'
/usr/bin/ld: /home/zetlin/Code/github/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so: undefined reference to `google::protobuf::internal::EpsCopyInputStream::DoneFallback(char const*, int)'
/usr/bin/ld: /home/zetlin/Code/github/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so: undefined reference to `google::protobuf::internal::VerifyUTF8(google::protobuf::StringPiece, char const*)'
/usr/bin/ld: /home/zetlin/Code/github/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so: undefined reference to `google::protobuf::RepeatedField<float>::Add(float const&)'
/usr/bin/ld: /home/zetlin/Code/github/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so: undefined reference to `google::protobuf::internal::ReadSizeFallback(char const*, unsigned int)'
/usr/bin/ld: /home/zetlin/Code/github/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so: undefined reference to `google::protobuf::RepeatedPtrField<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::Add()'
/usr/bin/ld: /home/zetlin/Code/github/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so: undefined reference to `google::protobuf::internal::WireFormat::InternalSerializeUnknownFieldsToArray(google::protobuf::UnknownFieldSet const&, unsigned char*, google::protobuf::io::EpsCopyOutputStream*)'
/usr/bin/ld: /home/zetlin/Code/github/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so: undefined reference to `google::protobuf::internal::UnknownFieldParse(unsigned long, google::protobuf::UnknownFieldSet*, char const*, google::protobuf::internal::ParseContext*)'
/usr/bin/ld: /home/zetlin/Code/github/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so: undefined reference to `google::protobuf::RepeatedPtrField<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::RepeatedPtrField(google::protobuf::Arena*)'
[... cut ...]
make[2]: *** [CMakeFiles/bevfusion.dir/build.make:186: bevfusion] Error 1
make[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/bevfusion.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

```
libprotobuf-dev version
```
(base) ➜  CUDA-BEVFusion git:(master) ✗ sudo apt list | grep "libprotobuf-dev"
[sudo] password for zetlin: 

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libprotobuf-dev/focal-updates,focal-security,now 3.6.1.3-2ubuntu5.2 amd64 [installed]
libprotobuf-dev/focal-updates,focal-security 3.6.1.3-2ubuntu5.2 i386

```

chjej202, 2024-06-13:
I also need spconv 1.1.2 with cuda 12.
I cannot use pyscn on Orin board  with Jetpack 6.0

hopef, 2024-06-24:
Sorry. I have pushed the CUDA 12 version of the libspconv. [link](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/libraries/3DSparseConvolution/libspconv_cuda12)

hopef, 2024-06-24:
**however the results are off:**
These issues may caused by layernorm layer in TensorRT.
I have pushed the layernorm plugin [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/src/plugins/custom_layernorm.cu).

1punch3coins, 2024-08-17:
@hopef Hey, the libpconv from https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/libraries/3DSparseConvolution/libspconv_cuda12 could not be used directly in your CenterPoint implementation, as some apis such as spconv::load_engine_from_onnx() and forward() are not provided.

hopef, 2024-08-20:
Sorry, I will update the CenterPoint implementation.

hyxhope, 2024-12-25:
> Sorry, I will update the CenterPoint implementation.

Any Updates? Sir

leoxiang66, 2025-03-19:
> Sorry, I will update the CenterPoint implementation.

Hi, I meet the same problem: spconv::load_engine_from_onnx() is not provided. Have you fixed it?

leoxiang66, 2025-03-19:
> [@hopef](https://github.com/hopef) Hey, the libpconv from https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/libraries/3DSparseConvolution/libspconv_cuda12 could not be used directly in your CenterPoint implementation, as some apis such as spconv::load_engine_from_onnx() is not provided and forward() is not compatible.

Hi, did you solve it?

Tigerold, 2025-03-20:
@xuyuan Hello, I have encountered the save issue, Have you solved it yet? 
I used two methods, the first one was to use deepstream branch, the second was move the spconv.so from deepstream branch to the code in the master branch, both of them was the same result as your picture.