NVIDIA-AI-IOT/Lidar_AI_Solution #235: spconv/engine.hpp: No such file or directory

k-nxt, 2024-01-31:
Hi, 

I tried to run CUDA-BEVFusion@8cd961f35660cb886edfd430015ef5de12ff3486.

When I execute `bash tool/run.sh`, I got this error:

>/Lidar_AI_Solution/CUDA-BEVFusion/src/bevfusion/lidar-scn.cpp:26:10: fatal error: spconv/engine.hpp: No such file or directory
   26 | #include <spconv/engine.hpp>

It seems there is no engine.hpp in **bevfusion/mmdet3d/ops/spconv/include/spconv/**

`ls bevfusion/mmdet3d/ops/spconv/include/spconv/`
>fused_spconv_ops.h  indice.h     point2voxel.h    reordering.h
geometry.h          maxpool.h    pool_ops.h       spconv_ops.h
indice.cu.h         mp_helper.h  reordering.cu.h

My environment:
- nvidia/cuda:11.3.1-devel-ubuntu20.04
- cuda-12.1
- environment.sh  
```
export TensorRT_Lib=/usr/src/tensorrt/lib
export TensorRT_Inc=/usr/src/tensorrt/include
export TensorRT_Bin=/usr/src/tensorrt/bin

export CUDA_Lib=/usr/local/cuda-12.1/lib64
export CUDA_Inc=/usr/local/cuda-12.1/include
export CUDA_Bin=/usr/local/cuda-12.1/bin

export CUDA_HOME=/usr/local/cuda-12.1
export CUDNN_Lib=/usr/local/cuda-12.1/lib64
```

How can I solve this?

Thanks in advance.

hopef, 2024-01-31:
Sorry, the default configuration is based on cuda11x, for cuda12, some modifications are needed. Please refer to the comments.
For this error, it may be caused by wrong dependency. **libspconv.hpp** it is configured to **3DSparseConvolution** in the parent directory. Please check if the directory exists.

k-nxt, 2024-02-01:
OK, thanks!

k-nxt, 2024-02-01:
That makes sense!

closed 2024-02-01 by k-nxt

IdleIdiot, 2025-05-26:
I’m having the same issue. What should I do?