NVIDIA-AI-IOT/Lidar_AI_Solution #236: CMake Compile Error

JYS997760473, 2024-02-05:
# Cannot compile the project
- Hardware: RTX4060
```
==========================================================
||  MODEL: resnet50int8
||  PRECISION: int8
||  DATA: example-data
||  USEPython: OFF
||
||  TensorRT: /usr/lib/x86_64-linux-gnu
||  CUDA: /usr/local/cuda-11.8
||  CUDNN: /usr/lib/x86_64-linux-gnu
==========================================================
Try to get the current device SM
Current CUDA SM: 89
Configuration done!
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-11.8 (found version "11.8") 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (found version "3.12.4") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jiayansong/workspace/Lidar_AI_Solution/CUDA-BEVFusion/build
[  4%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/common/bevfusion_core_generated_tensor.cu.o
[ 13%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-bevpool.cu.o
[ 13%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-backbone.cu.o
[ 18%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-depth.cu.o
[ 22%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-geometry.cu.o
[ 27%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-normalization.cu.o
[ 31%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_transfusion.cu.o
[ 36%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-vtransform.cu.o
[ 40%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_head-transbbox.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_lidar-voxelization.cu.o
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
CMake Error at bevfusion_core_generated_camera-depth.cu.o.Release.cmake:280 (message):
  Error generating file
  /home/jiayansong/workspace/Lidar_AI_Solution/CUDA-BEVFusion/build/CMakeFiles/bevfusion_core.dir/src/bevfusion/./bevfusion_core_generated_camera-depth.cu.o

make[2]: *** [CMakeFiles/bevfusion_core.dir/build.make:91: CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-depth.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMake Error at bevfusion_core_generated_lidar-voxelization.cu.o.Release.cmake:280 (message):
  Error generating file
  /home/jiayansong/workspace/Lidar_AI_Solution/CUDA-BEVFusion/build/CMakeFiles/bevfusion_core.dir/src/bevfusion/./bevfusion_core_generated_lidar-voxelization.cu.o

```

JQZhai, 2024-02-23:
I have a similar problem, how to solve it？

JYS997760473, 2024-02-23:
> I have a similar problem, how to solve it？
I changed to use docker instead, then can build now. My host is Ubuntu 22.04, docker is ubuntu 20.04. Maybe you can try docker

hopef, 2024-06-24:
Oh, it looks like you are compiling in an abnormal environment.

anteaglew2hcs, 2024-09-13:
I got similar issue as well, has anyone fixed this ?

hopef, 2024-09-14:
Sorry, this is caused by the option -std. 
For CUDA 11.x, we should set the option to -std=c++14. 
For CUDA 12.x, the option should be set to -std=c++17.

closed 2024-09-15 by JYS997760473